mmattel/EET-Solmate #7: How to install/integrate the pyhton script in Home Assistant

BetaTobi, 2023-10-29:
Hi,

first of all: Many thanks for your work. It is exactly, what I need or want to have.

But a have an issue to get it running:

I am trying to integrate/install the python script in my Home Assistant.
Unfortunately, it does not work and I think I am doing something wrong.

What I did so far:

1: read the pyhton script info from HA: https://www.home-assistant.io/integrations/python_script/
2: downloaded all the files of the repo to my local home assistant directory: "...\config\python_scripts\EET-Solmate-main"
3: renamed ".env-sample" into ".env" and modified the content
4: restarted HA

Main questions, especially my point #2:
Is the downloading "enough" for the installation?
Or how can I install the python script and the required modules?
I am (more or less) familiar with the SSH terminal (if necessary), but I am lacking the knowledge how to install the pyhton script correctly

How can I start the pyhton script (if/when it is installed properly).

Greetz

mmattel, 2023-10-29:
@BetaTobi Thanks for adding more info.
* Your intention to let the script run as Python HA integration **can not** work and is easily explained by your link provided `...Each Python file created in the <config>/python_scripts/ folder will be exposed as a service.` My script contains a set of single python files working together and not as single one. I also doubt that making it a single script would work as the necessary error handling will in case restart the script which may negatively interfere with HA.
* Maybe not expressed explicitly in the Readme (which I will fix asap for clarification), but HA, MQTT and my Python script are independent units. You need to have as prerequisite HA and MQTT up and running successfully. They can therefore run on separate hosts/containers and connect to each other as configured.

To successfully run the script:
* Add/Clone the script to a folder of your choice, recommended to use your home directory.
* Check that my Python script has the necessary modules available.
* Configure the .env file.
* Open two shells (assuming you are running Ubuntu/Debian):
   * In the first shell run: `tail -f /var/log/syslog | grep solmate` to monitor logs.
Alternatively set `console_print` temporarily to true in the script.
   * In the second shell start the script
* You should now see the script running successfully. If not check the configuration.
* Monitor MQTT posts, use `MQTT Explorer`. The SolMate should show up.
* Check that HA shows in MQTT as source the new EET entry.

mmattel, 2023-10-29:
@BetaTobi FYI, I just have updated the readme to reflect the comments above.

BetaTobi, 2023-10-29:
Many thanks!
I will try this out and come back with a feedback (either positive or negative :-) )

BetaTobi, 2023-10-29:
Ok.. cloning worked well.
Then, I check for the missing requirements and got this:

[image]

I commented all the listed requirements in the "requirements.txt" and then typed "python -m pip install -r requirements.txt".

This was the output:

[image]

I checked again the requirements and got this:

[image]

What does this mean?
Are all modules now installed?

mmattel, 2023-10-29:
> Are all modules now installed?

Sieht danach aus weil keine fehlenden Packages gemeldet wurden. Das mit der "Fehlermeldung" check ich noch könnte P3.11 spezifisch sein.

Was für ein OS wird benutzt?

Welche Rückmeldungen gibt es wenn solmate.py gestartet wird (mit oder ohne config), denn Python prüft am Anfang ob alle Module installiert sind?

BetaTobi, 2023-10-29:
Als OS läuft bei mir direkt Home Assistant, auf einem RPi.
HA läuft also nicht als Container oder so.

Wenn ich "solmate.py" starte, erhalte ich die folgende Fehlermeldung:
[image]

Dann hat das installieren der Module wohl doch nicht funktioniert.
Kann ich die auch manuell einzeln installieren?

mmattel, 2023-10-29:
> Kann ich die auch manuell einzeln installieren?

Klar: `python -m pip install <package-name>` ohne die spitzen Klammern...

BetaTobi, 2023-10-29:
Ok, das war einfach.
Danke.

Ich konnte alle Module installieren, bis auf "dotenv".
Das gibt immer diese Fehlermeldung und "solmate.py" lässt dann auch nicht starten.

Hast du eine Idee?

[image]

mmattel, 2023-10-29:
Da müssen wir uns herantasten da Du HA-OS hast...
Probier bitte mal folgendes in dieser Reihenfolge, das nächste nur wenn das vorhergehende nicht funktioniert hat:

1. `python -m pip install python-dotenv`
2. `python -m pip install --upgrade setuptools` und dann
`python -m pip install dotenv` (bzw  `python-dotenv`)

Möglich dass `dotenv` nach `python-dotenv` umbenannt wurde.

BetaTobi, 2023-10-29:
Danke, das war es!

Also Option 1 ("pyhton_dotenv" anstatt "dotenv").

Und MQTT-Daten sind auch sofort da.
Ein Traum :-)

Noch einmal vielen Dank!
Und du brauchst dringend einen "buy me a coffee"-button :-)

Hier mal die Infos und Screenshots dazu:

[image]

[image]

mmattel, 2023-10-29:
Super, freut mich dass es bei dir nun klappt und danke für dein Feedback!
Ich mach gerade eine Verbesserung der `requirements` damit das auch smooth läuft.

Hier noch ein paar Hinweise 😄 
* Dein Solmate läuft auf Version 0.4.52-d - die ist aktueller als meine welche 4.48-d ist 😅 
* Lass Dich nicht durch die beiden Timestamps verwirren. Der erste ist die Life Abfrage und ändert sich alle 30s, der zweite immer dann wann das Script neu gestartet wurde (warum auch immer) ODER 1x am Tag in der Nacht weil die SW Version ändert sich nicht so häufig. Nachdem die Serverabfrage auf Seiten EET noch "Potental" an Stabilität hat, siehst Du hier auch wann sich das Script wegen nicht behebbarem Fehler neu starten musste. Das sollte ich noch in das Readme schreiben...
* Ich weis nicht wie das in HA-OS geht, aber bei mir läuft das Ganze mit systemd wunderbar. Damit startet das Ganze beim Reboot automatisch und Du kannst den Prozess auch sehr gut manuell steuern.
* Check auch von Zeit zu Zeit das syslog wie oben geschrieben. Ach, ich komm in's schwärmen 😄

mmattel, 2023-10-29:
Hab das Requirement handling upgedated. Bei Bedarf das `check_requirements.py` und `requirements.txt` file updaten. Das Readme hat auch noch einen Refresh im HA Bereich bekommen.

Wenn der Issue für Dich gelöst ist, bitte closen.
Bei neuen Themen einfach einen neuen aufmachen.

BetaTobi, 2023-10-29:
Alles klar, vielen Dank.

Wie ich das Script starte, wenn HA neu startet, hab ich noch nicht rausgefunden, aber da find ich schon was.
Hauptsache das Script läuft, und das tut es ja.

Ich mach dann hier zu.

closed 2023-10-29 by BetaTobi

mmattel, 2023-10-29:
> Wie ich das Script starte, wenn HA neu startet...

**Info:**
* Das ist UNABHÄNGIG 😄 
* Das Script muss starten wenn der Server neu startet zB Reboot, Power Cycle etc.
Als Abhängigkeit kann man zb in sytemd ein Service eintragen das zuvor gestartet sein muß, siehe Beispiel: [Setup Docker to Wait for NFS Mounts](https://github.com/mmattel/Raspberry-Pi-Setup/tree/main#setup-docker-to-wait-for-nfs-mounts) wegen dem Verfahren, zb der MQTT broker der online sie muß.

BetaTobi, 2023-10-30:
> * Das ist UNABHÄNGIG

Ja, hab mich schlecht ausgedrückt.
Ich meinte, wenn ich den ganzen Computer (RPi) neustarte (auf dem HA läuft).
Dann muss ich natürlich auch das Script neu starten. Ich schau mir "systemd" mal an.

Aber ich hab ein anderes Problem festgestellt:
Wenn ich den kompletten RPi neu starte, dann sind auch die Pyhton Module (websockets, schedule, python-dotenv...) "weg".
Ich muss diese neu installieren und dann erst kann ich solmate.py starten.
Sehr merkwürdig..

mmattel, 2023-10-30:
> Wenn ich den kompletten RPi neu starte, dann sind auch die Pyhton Module weg.

Oh mann,
HA-OS ist schon mühsam...
Genau deswegen (befürchtet) hab ich es nicht genommen und auf meinem RPi Debian und Docker installiert, siehe mein anderes Repo.

An einer Info zu einer Lösung bin ich interessiert, das würde ich im Readme zusätzlich dokumentieren.

Info: arbeite gerade an einem Upgrade mit dem man dann den solmate über HA/MQTT rebooten kann. Meiner "verliert" alle paar Wochen die Bereitstellung der PV Leistung und zeigt bei vollem Tageslicht 0W Paneel an. Ein Reboot tuts dann.

mmattel, 2023-10-30:
Möglicherweise [braucht man das](https://github.com/AlexxIT/PythonScriptsPro) und schlägt zwei Fliegen mit einer Klappe?

BetaTobi, 2023-10-30:
Ich hab gestern mal [pyscript](https://github.com/custom-components/pyscript) installiert, aber bin noch nicht weiter zum testen gekommen.
Werd in den nächten Tagen wohl auch nicht mehr dazu kommen.
Aber sobald ich eine Lösung gefunden hab (oder neue Fragen hab :) ), schreib ich es hier.

lukashunziker99, 2023-11-26:
Hallo Ihr Beiden
Super Arbeit.
Ich habe das gleiche System wie BetaTobi.
Alle requirements sind installiert und ich kann solmate.py starten. Folgende Meldung kommt dann:

➜  EET-Solmate python solmate.py
Initializing websocket connection...
Authenticating...
Websocket is connected to: wss://sol.eet.energy:9124/
Response: {'type': 'DoesNotExist', 'message': ''}
Authentication failed!
Failed creating connection/authentication to websocket class.
Waiting: timer_offline: 600s

Im .env habe ich meine Solmate IP so eingegeben.
eet_server_uri="ws://sun2plug.192.168.1.55:9124/"

Solmate ist dabei online.

Was mache ich falsch?
Herzlichen Dank für ein feedback.

mmattel, 2023-11-26:
Eine Bitte, mach einen neuen Issue auf, ich weis das macht Arbeit, aber dann kann ich darauf besser eingehen und die Dinge trennen. Ich antworte gerne wenn dein Issue da ist 😄