mmattel/EET-Solmate #11: Authentication failed

lukashunziker99, 2023-11-26:
Hallo Ihr Beiden
Super Arbeit.
Ich habe das gleiche System wie BetaTobi.
Alle requirements sind installiert und ich kann solmate.py starten. Folgende Meldung kommt dann:

➜ EET-Solmate python solmate.py
Initializing websocket connection...
Authenticating...
Websocket is connected to: wss://sol.eet.energy:9124/
Response: {'type': 'DoesNotExist', 'message': ''}
Authentication failed!
Failed creating connection/authentication to websocket class.
Waiting: timer_offline: 600s

Im .env habe ich meine Solmate IP so eingegeben.
eet_server_uri="ws://sun2plug.192.168.1.55:9124/"

Solmate ist dabei online.

Was mache ich falsch?
Herzlichen Dank für ein feedback

BetaTobi, 2023-11-26:
Hi @lukashunziker99 

ich hab im .env-file nur das hier stehen:

eet_server_uri="wss://sol.eet.energy:9124/"

--> also ohne zusätzliche Angabe der lokalen IP (die kommt eh von sol.eet.energy).
Probier es mal aus.

Viele Grüße

lukashunziker99, 2023-11-26:
Danke für die Antwort. Dh Du hast das Originaleintrag  .env-sample belassen:
# solmate config
# get the local domain from your network setup (router, dhcp etc.)
#eet_server_uri="ws://sun2plug.<your-domain>:9124/"

eet_server_uri="wss://sol.eet.energy:9124/"

Wenn ich es so mache kommt bei der gleiche Fehler wie bereits oben beschrieben. 

Bei:
 eet_serial_number="<solmate-serial-number>"
eet_password="<solmate-password>"
eet_device_id="<solmate-given-name>"

habe ich die korrekten Angaben ohne <> eingegeben.

BetaTobi, 2023-11-26:
Gern 👍 😊

mmattel, 2023-11-26:
@lukashunziker99 @BetaTobi mitte Dezember kommt, wenn alles gut geht eine neue Release. Ist ein großes Refactoring das viel positives bringt wie viel selteneres durchstarten oder die Möglichkeit den Solmate so wie im WebUI auch über MQTT/HA zu rebooten. Das Readme wird dann alles beschreiben. Stay tuned 😄

BetaTobi, 2023-11-26:
Okay, klingt gut :-)
Dann warte ich mit meinen Problemchen bis dahin 😉

lukashunziker99, 2023-11-26:
Super - freue mich.

mmattel, 2023-12-01:
@BetaTobi @lukashunziker99 Good news, die neue Release v2.0.0 ist da und kann verwendet werden.
Bitte UNBEDINGT das README lesen da Major = es gibt Breaking Changes...
Upgrade ist aber einfach und kein Problem 😄 
Feedback welcomed. Alles andere in einem eigenen Issue!

closed 2023-12-01 by mmattel