mmattel/EET-Solmate #17: Solmate offline obwohl online

lukashunziker99, 2023-12-02:
Hallo
Danke. Folgendes Problem.
Solmate erscheint offline obwohl über Browser mit IP online.
Folgende Einstellungen in .env:
# use local access if you also want to reboot the solmate
# get the local domain from your network setup (router, dhcp etc.)
# note: MUTUAL EXCLUSIVE to the internet eet_server_uri entry. only one must be enabled
eet_server_uri="ws://192.168.1.55:9124/"

# subdomain for the local solmate dns name, check your (routers) dns entry for the solmate
# MUST be populated when using local access. not needed when using the server below
eet_local_subdomain="sun2plug"

# needs an internet connection
# note: MUTUAL EXCLUSIVE to the local eet_server_uri entry. only one must be enabled
#eet_server_uri="wss://sol.eet.energy:9124/"

falls ich:  eet_server_uri="ws://sun2plug.192.168.1.55:9124/" kommt Websockets error: [errno -2] Name does not resolve

Was mache ich falsch?

mmattel, 2023-12-02:
> ws://sun2plug.192.168.1.55:9124/

Na ja, die URI ist einfach falsch, Du mischt die Nomenklatur...

**Entweder 1.:** ws://192.168.1.55:9124/
**oder 2.:**
ws://sun2plug.<deine-lokale-domain>:9124/

mmattel, 2023-12-02:
Kannst du mal bitte in einer shell folgendes eingeben und das Ergebnis hier posten?
`nslookup 192.168.1.55`

mmattel, 2023-12-02:
Ok, die Lösung ist einfach:
1. Ich hab mehr Kommentar in die `.env-sample` geschrieben um das aufzuklären.
2. Für Dich, gib in die `eet_server_uri` folgendes ein: `ws://sun2plug.home:9124/`

Du kannst das einfach testen, gib dazu folgendes ein: `ping sun2plug.home` und als Ergebnis solltest Du `192.168.1.55` bekommen. Hinweis, natürlich kann man eine IP Adresse verwenden aber das ist nie garantiert dass die sich nicht ändert, eine FQDN ist statisch und deshalb bevorzugt.

Bitt eum kurze Meldung wenn sich der Solmate damit online meldet.

lukashunziker99, 2023-12-02:
Great - works now with ws://sun2plug.home.:9124/

Thanks

mmattel, 2023-12-02:
Closing because of success 😄

closed 2023-12-02 by mmattel

lukashunziker99, 2023-12-02:
Funktioniert jetzt so mit FQDN - super. Ich habe aber eine statische IP fürs Solmate und verstehe es nicht ganz. Herzlichen Dank.

mmattel, 2023-12-02:
Der FQDN ändert sich defacto nie, außer EET ändert den vorgeschlagenen Hostname vom Solmate für die Autodiscovery im DNS oder wenn Du den Hersteller Deines Routers wechselst. Der FQD ist ebenfalls fix wenn Du die ganzen DNS Einträge und den Solmate manuell einrichtest. Die IP Adresse kannst Du aber jederzeit anpassen oder mit DHCP vergeben lassen und im letzteren Fall kann die sich ändern. Um das zu vermeiden ist FQDN einfach viel besser weil Du dann weist dass Du immer auf ihn zugreifen kannst. Teste mal in der Weboberfläche vom Solmate anstatt der IP den FQDN 😄 Ehrlich, willst Du Dir immer die IP Adresse merken?

PR #18 (2023-12-02): Clarify in .env-sample the use of eet_local_subdomain
Fixes: #17 (Solmate offline obwohl online)

The value of `eet_server_uri` when using local access and `eet_local_subdomain` and in `.env-sample` needs clarification to get a FQDN respectively when using an IP address instead a FQDN.