mmattel/EET-Solmate #20: Autostart auf Home Assistant Green

MadNasedo, 2023-12-08:
Hallo,
ich hab hier HomeAssistant auf einem Home Assistant Green laufen.
Das Script ist installiert und läuft auch wenn ich es vom Terminal aus starte - Daten kommen in MQTT an und ich hab mir auch schon  aber das die Solmate.py automatisch gestartet wird bekomme ich einfach nicht hin.

Der Befehl:
`sudo systemctl edit --force --full eet.solmate`
ergibt hier nur:
`sudo: systemctl: command not found`

Hatte HomeAssistant vorher auf einem Intel Nuc laufen, aber da wars das gleiche.

Wie bekomme ich den Autostart noch hin? Hast du da vielleicht eine Idee?
Was ist der Sensor.mppOutI ?

mmattel, 2023-12-08:
Hier eine Erklärung zu [mpp](https://de.wikipedia.org/wiki/Maximum_Power_Point_Tracking?wprov=sfti1)

Wegen systemd, Du verwendest HASSIO, also ein OS das von HA "downgesized" wurde. Wie ein Autostart in HASSIO bewerkstelligt werden kann weis ich nicht. Da musst schauen, freue mich auf was Du findest. Wenn ich was finde schreib ich.

MadNasedo, 2023-12-09:
So, alles noch mal von neuem installiert - jetzt läuft gar nichts mehr :(
```
 EET-Solmate git:(main) ✗ python solmate.py
Traceback (most recent call last):
  File "/homeassistant/EET-Solmate/solmate.py", line 7, in <module>
    import schedule
ModuleNotFoundError: No module named 'schedule'
```
Ich denke HASSIO ist zu minimal, um das Script vernünftig zu betreiben.

mmattel, 2023-12-09:
Sind die Requirements erfüllt? sind alle module vorhanden? Hast du das check-requirements.py script ausgeführt?

MadNasedo, 2023-12-09:
Ja, die Requirements sind laut Script erfüllt.
` python check_requirements.py
/usr/lib/python3.11/site-packages/_distutils_hack/__init__.py:33: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")

The following modules can be commented in requirements.txt because they are already available:

[]

When done, call: python -m pip install -r requirements.txt

If the call returns an empty array like [], all module requirements are met.
`

mmattel, 2023-12-09:
Alles der Reihe nach...

KeyError: 'general_add_log'

bedeutet, dass der key in der .env Datei nicht gefunden wurde. Check mal ob in der .env alle keys vom sample drinnen und konfiguriert sind.

MadNasedo, 2023-12-09:
Die .env war das Problem - ich habe einfach die von der älteren Version gelassen. Alle Werte in die neuere übertragen -> läuft wieder :D 

Jetzt muss ich nur noch rausfinden wie ich den Autostart hinbekomme.

Danke 👍

mmattel, 2023-12-09:
Sehr gut. Check mal ob in HASSIO systemd vorhanden ist,
zb über https://linuxhandbook.com/check-if-systemd/amp/
bzw gibt es einen folder /etc/system/systemd

MadNasedo, 2023-12-09:
Hassio benutzt systemd nicht, es wird busybox benützt.
```
➜  ~ stat /sbin/init
  File: '/sbin/init' -> '/bin/busybox'
```

BetaTobi, 2023-12-09:
#

> Jetzt muss ich nur noch rausfinden wie ich den Autostart hinbekomme.

Bei dem Problem steh ich auch.
Script läuft, aber nach jedem Neustart muss ich einige Python Module neu installieren und das Script auch selbst starten.

Ich such auch nach einer Möglichkeit die nötigen Terminbefehle z.B. via Automation zu senden.
Aber bisher wenig erfolgreich.
Weiß jemand, ob das geht?
Und falls ja, wie?

:-)

MadNasedo, 2023-12-09:
Same here... sobald in Putty beende oder das Advanced Terminal Addon neustarte, sind auch ein paar Module weg.

mmattel, 2023-12-09:
Möglicherweise wäre das ein Weg:

https://smarterkram.de/2524/

* Dann bleiben schon einmal die Module da.
* Dann müsste man noch testen ob man mein script damit starten kann.
* Und final müsste man einen schwerwiegenden fehler provozieren um zu sehen ob das Ganze einen selbst getriggerten full script restart überlebt. Dazu könnte man zb einen counter in der Hauptloop in solmate.py einbauen der den reset triggert...

Auch wenn der Betrieb mit RPi und Docker etwas aufwendiger ist, siehe mein anderes Repo, schätze ich die Freiheiten damit sehr!

MadNasedo, 2023-12-09:
Hab das mal installiert und in die configuration.yaml das reingeschrieben:
```
python_script:
    requirements:
        - asyncio
        - datetime
        - paho-mqtt
        - python-dotenv
        - schedule
        - termcolor
        - websockets   
script:
  solmate:
    sequence:
    - service: python_script.exec
      data_template:  # use `data_template` if you have Jinja2 templates in params
        file: homeassistant/EET-Solmate/solmate.py  # relative path from config folder
        cache: false  # disable cache if you want change python file without reload HA
        title: Solmate Python Script
        time_val: "{{ states('sensor.start_time')|round }}"
```
Aber es werden weder die Module geladen noch das Script gestartet.

mmattel, 2023-12-09:
Versuche mal, a la Windows, HA neu zu starten.

MadNasedo, 2023-12-09:
Leider ändert ein Kaltstart daran auch nichts, man sieht auch in den Protokollen nichts davon - ist halt die Frage ob das überhaupt in irgendwelchen Logs auftaucht.

BetaTobi, 2023-12-10:
Ich hab "Python Scripts Pro" auch installiert und getestet.
Bei mir geht es auch nicht.

Es werden weder die Module installiert, noch kann ich das Script starten (selbst dann nicht, wenn ich die Module vorher noch einmal manuell installier hab).

@MadNasedo 
Du könntest testweise mal den Pfad zum script ändern.
Bei mir ging nur "file: EET-Solmate/solmate.py" fehlerfrei.
Aber das Script wurd trotzdem nicht gestartet...

MadNasedo, 2023-12-10:
Hab ich gemacht, das Ergebnis ist leider das gleiche.
Im Advanced SSH & Web Terminal Addon kann ich dann die Module nachladen, erst dann läuft auch die Solmate.py

mmattel, 2023-12-10:
Ich werd mal, muss noch einen Zeitslot suchen, eine VM mit HASSIO installieren und nach Möglichkeiten schauen. Dauert aber etwas

MadNasedo, 2023-12-10:
Hat keinen Stress - momentan ists halt sehr interessant wieviel Temperatur der Solmate hat. Unter 4°C gefällt ihm nicht besonders.

BetaTobi, 2023-12-10:
> Hat keinen Stress - momentan ists halt sehr interessant wieviel Temperatur der Solmate hat. Unter 4°C gefällt ihm nicht besonders.

Hast du Solmate V1 oder V2?
Beim V2 ist eine Heizung eingebaut (wurd mir jedenfalls gesagt)

mmattel, 2023-12-10:
V2, im Frühsommer geliefert. SW Release 0.4.55-r. Funktioniert auch bei -7 Grad einwandfrei und wenn über Tage hinweg die Paneele voll Schnee sind. Zwar keine Einspeisung, aber die CPU bleibt Aktiv.

BetaTobi, 2023-12-10:
Eben, bei mir (auch V2, erst jetzt im Herbst geliefert).
Und der Akku wird auch geladen, wenn die Temperatur unter 0 war.
Deswegen frag ich mich auch, wo genau der Temperatursensor genau sitzt. 
Die Temperatur vom Akku selbst sollte ja beim Laden möglichst über 0° sein..

MadNasedo, 2023-12-10:
Ich hab den Solmate im April 2022 reserviert und Ende Juli 2023 bekommen. Ich denke es ist irgendwie V2 - man kann max. 4 Panele anschließen, wobei ich 3 Panele angeschlossen habe und fast südlich ausgerichtet bin. Es fällt auch der Akku öfter auf 0% runter obwohl 10% Restladung eingestellt sind. Das passiert meistens wenns sehr kalt ist - ich hab dem Solmate schon 3cm Aluschaumplatten rundherum gestellt.
[image]
Das System als ganzes funktioniert meiner Meinung nicht richtig - sobald der Akku voll ist, wird sofort die Einspeiseleistung zurückgefahren. Auf Nachfrage bei EET wurde mir nur gesagt das das Verhalten normal ist und mein Solmate ok ist. Auf meinen Einwand das diese Voodoo Messung, die der Solmate macht, um einen Lastwechsel zu erkennen niemals funktionieren kann, wurde nichteinmal reagiert :)  Ich hab einen SmartPi in der Zuleitung hängen und auch sonst auf jedem Abgang einen Shelly hängen, daher weiß ich das diese Aussage nur Marketing ist. Noch dazu hängt der Solmate auf einer eigenen Leitung :D
[image]

Auf meinem Solmate ist Version 0.4.46-d , und das schon sehr lange. Ein manuelles Update kann man da ja nicht anstoßen.

mmattel, 2023-12-10:
Auf dem SW Sektor hat EET gelinde gesagt noch sehr viel Potential. Deshalb hab ich auch eine eigene geschrieben die sich in diesem Repo wiederspiegelt... All das hier gehört in den Solmate. Die Update Politik ist grundlegend falsch. Das Wlan ist ein Witz, mein Solmate ist per USB-LAN verkabelt etc.
Aber genug des Meckerns, er funktioniert soweit und ich bekomm nun die Daten die ich brauche.

BetaTobi, 2023-12-10:
Bei mir läuft das System übrigens sehr gut.
Hab den Solmate diesen Herbst bekommen.
Der minimale Ladezustand wird respektiert und es wird nichts mehr eingespeist, wenn diese untere Grenze erreicht ist.
Auch der Bereich minimale bis maximale Einspeisung wird eingehalten.
Und selbst der WLan-Empfang ist gut und das System läuft stabil.
Ich kann da wirklich überhaupt nicht meckern.
Über die Stromverbrauchsmessung-Technology kann ich allerdings auch nichts sagen. Es hängen auch quasi keine wirklichen Verbraucher an derselben Phase.

Naja.. bleiben wir optimistisch :-)

mmattel, 2024-02-26:
Hier eine Zusammenfassung meiner Findings und ein paar Erklärungen:

1. Das Script läuft mega stabil, es gibt seit gestern eine v3 da die paho-mqtt library updatet wurde. (Siehe readme)
2. Die Installation auf einem System auf das man Kontrolle hat, zb HA läuft unter Docker ist problemlos. Alles Andere, HA Appliance oder HAOS ist mit Handarbeit verbunden da HA MASSIVE Einschränkungen macht. zB HA Shell Commands terminieren hart nach 60s etc. Das geht allerdings nicht nur mir so...
3. Eine Einbindung in HACS werde ich nicht machen. Dazu ist das Script zu komplex und meine Zeit zu knapp. HACS ist für einfache Dinge da.

Hoff das hilft euch weiter und ihr seid mit meiner Arbeit zufrieden.

closed 2024-02-26 by mmattel