mmattel/EET-Solmate #21: Upgrading from release 1.x to 2.x

BetaTobi, 2023-12-10:
Hi!

Ich bin aktuell noch auch Version 1.x und möchte auf die neue Version 2.x updaten.
Die readme dazu hab ich gelesen, aber dennoch eine Frage.

Muss ich die alte Version 1.x vorher lokal löschen?
Oder kann ich die neue Version 2.x einfach "drüberkopieren"?

Ansonsten würd ich einfach das neue Repo clonen, mit dem Befehl:
"gh repo clone mmattel/EET-Solmate"

Und anschließend das .env-file anpassen.

Ist das so richtig?

Viele Grüße

mmattel, 2023-12-10:
Hallo @BetaTobi, einfach drüber clonen reicht. Danach am besten ein neues .env vom sample erstellen und auf Deine Bedürfnisse anpassen.

BetaTobi, 2023-12-10:
Hi @mmattel 
Danke, hat funktioniert.

Allerdings nicht mit "git pull" --> das gab einen Fehler ("fatal: not a git repository").

Mit "git clone" hat es ZUNÄCHST auch nicht funktioniert, da das Zielverzeichnis bereits existiert.
Erst nachdem ich das komplette Verzeichnis gelöscht hatte, hat es funktioniert.
Nur löschen des ".git"-Verzeichnisses hat (bei mir) auch nicht gereicht.

Also ich hab es so gemacht:

- alte .env-Datei kopiert/gesichert
- komplettes "EET-Solmate"-Verzeichnis gelöscht
- neues Git-Repo gecloned (mit Befehl: "git clone https://github.com/mmattel/EET-Solmate.git"
- neue .env-Datei erstellt/editiert

Läuft :)

Ich bin auch aktuell lokal verbunden (lokale IP).

Danke!

closed 2023-12-10 by BetaTobi

mmattel, 2024-05-04:
Info:
https://github.com/mmattel/EET-Solmate/releases/tag/v5.0.0
https://github.com/mmattel/EET-Solmate/discussions