mmattel/EET-Solmate #22: Initializing the MQTT Client Error nach core_2024.2.1 update

lukashunziker99, 2024-02-19:
Seit dem Core2024.2.1 update erhalte ich Fehler beimInitialisieren des MQTT Clients (siehe Bild). Kann das jemand nachvollziehen. MQTT funktioniert grundsätzlich normal, nur mit SolMate nicht mehr. Danke für Hilfe.
[image]

lukashunziker99, 2024-02-24:
Problem gelöst:. Es lag am PAHO MQTT mit der neuen 2.0.0 Version:
Im File solmate_mqtt.py musste ich Zeile 67 so ändern: self.mqttclient = mqtt.Client(mqtt.CallbackAPIVersion.VERSION1, client_id = self.mqtt_client_id, clean_session = True)

BetaTobi, 2024-02-24:
Gleicher Fehler bei mir von HA Core 2024.2.2 auf HA Core 2024.2.3.

Ist reproduzierbar.

@lukashunziker99 Deine Lösung probier ich gleich aus.
Hab grad eben wieder das Backup auf die "funktionierende" HA Core Version zurückgespielt und muss jetzt wieder updaten :-)

[image]

BetaTobi, 2024-02-24:
So, Update aufgespielt und die Änderung von @lukashunziker99 eingebaut --> und es läuft wieder.
Besten Dank!

lukashunziker99, 2024-02-24:
Super. Es liegt aber nicht am HA Core update, sondern an der neuen PAHO MQTT Version. Du kannst problemlos die HA Core Updates installieren.

mmattel, 2024-02-25:
Also...

Es gibt eine neue Version - v3 😄 
Die behebt das Problem grundlegend und macht keine backwards Kompatibilität sondern geht korrekt auf v2 der `paho-mqtt` library und ist damit zukunftssicher. Lest dazu das README, es enthält alle Informationen (Breaking und Improvements). Ich hab btw alle meine lokalen Libraries upgedatet und konnte nichts negatives feststellen.

Viel Spass!

closed 2024-02-25 by mmattel