mmattel/EET-Solmate #25: Tests if libraries survive a reboot when using a Python virtual env on HAOS

mmattel, 2024-02-27:
@lukashunziker99 @BetaTobi 

To relax the `intalled libraries are gone`  on a HAOS system post a reboot, the following would be interresting to be checked. I dont use HAOS and must rely on community feedback.

**Procedure:**
* Create a _new_ directory of choice like `mkdir /test`
* Run `python -m venv /test`
* Run `cd /test`
* Run `source /test/bin/activate`
* Run `python -m pip freeze | grep paho`
* Check that this library is NOT installed
* Run `python -m pip install paho-mqtt`
* Reboot you HAOS server
* Run `cd /test`
* Run `source /test/bin/activate`
* Run `python -m pip freeze | grep paho`
* Check the output for the precense of the formerly installed library and post the result here.

If the library is present, it is reboot persistent in a python virtual env and that is easier to handle on a HAOS reboot as you only need to install the required libraries once.

lukashunziker99, 2024-02-27:
Thank you. I tried it. After reboot the directory was gone - see photo (I used luk instead of test). Sadly this does not work.
[image]
Is there a possibility to add an adjustment for {client.set_min_injection(200)} , {client.set_max_injection(800)} and minimal battery load - sorry I am not capable to write this.

mmattel, 2024-02-27:
@lukashunziker99 Thanks for the info.
Could you do the same test in /config/<your dir> ?

josy1024, 2024-02-28:
> Is there a possibility to add an adjustment for {client.set_min_injection(200)} , {client.set_max_injection(800)} and minimal battery load - sorry I am not capable to write this.

@lukashunziker99 I've running my custom code, because i want to dynamically set min, max and battery_percentage with mqtt.

the sol2mqtt.py script proxys the values to mqtt and subscribes from mqtt,
i'm using node-red integration in home-assistant for changeing the mqtt values.

config-data flows as follows:
* **node-red** function (checks grid values)
** if power flows to grid => load battery (min, max = 0, battery percentage = 95)
** if power flows from grid => use battery (battery_percentage = 20, min, max depends on grid usage between 50-200 watts)
* * => if something changed => write to "subscribe_topics" MQTT
* **sol2mqtt.py** subscribed "subscribe_topics" from MQTT => writes back with official "solclient" (solmate_sdk)

It's working, long-time tested and robust although solmate api is not very stable and the script needs resets every 1-2 hours.
maybe @mmattel could integrate this and use some of my code after ensureing your coding standards.
i could also do some tests in my HAOS installation if you want? 

just need the https://github.com/eet-energy/solmate-sdk running in your env, then you can use this script.
(

https://github.com/josy1024/eet2mqtt/blob/main/sol2mqtt.py
searchfor "subscribe_topics" and "mqtt2sol":
```py

subscribe_topics = ["eet/solmate/0/set/user_maximum_injection", "eet/solmate/0/set/user_minimum_injection", "eet/solmate/0/set/user_minimum_battery_percentage"]

def mqtt2sol(topic, received_message):
    print(f"mqtt2sol: Received message on topic solclient.set {topic}: {received_message}")
    if "user_maximum_injection" in topic:
        solclient.set_max_injection(int(received_message))
    elif "user_minimum_injection" in topic:
        solclient.set_min_injection(int(received_message))
    elif "user_minimum_battery_percentage" in topic:
        solclient.set_min_battery_percentage(int(received_message))
```

lukashunziker99, 2024-02-28:
Did it in /config. Sadly same result.
[image0.jpeg]

mmattel, 2024-02-28:
@josy1024 could you pls not "highjack" this issue for other topics and create a new one to separate content. Pls open a new issue for your request. This issue is about something totally different 😄

mmattel, 2024-02-28:
> Did it in /config. Sadly same result.

(Note, your image with this comment is not visible)

Hmmmm,
* You proofed that stuff outside the `/content` directory will go away on reboot - I expected this.
* But for stuff inside the `/content` directory, I did not expect this and this looks strange. Everthing in the `/content` directory should be reboot persistent, else nothing like settings or configs would be persistent. At least on my side, I am running HA in docker, all inside `/content` persists including directories und stuff created inside. I have created a python virtual env in `/content/test` and it was reboot persistent without any issue. This is finally the thing I want to check if it behaves the same in HAOS.

I am wondering btw where you have put your solmate scripts because if not in `/content/<your location>`, it would be gone on any reboot...

lukashunziker99, 2024-02-28:
I think all content in virtual env is deleted upon reboot. Usual folders can only be opened without /test (cd test not cd /test). [image0.png]

josy1024, 2024-02-28:
last year i'v done some testing with appdaemon and pyscript concluding a seperate rpi4 zero with custom code works in no time without headaches. 🤯

resumee:
* you can use and preload modules in custum py scripts. (yes, that works)
* all code must explicity written for home-assistant and async-io. (solmate_sdk isn't!)

* [https://appdaemon.readthedocs.io/en/latest/HASS_TUTORIAL.html](https://appdaemon.readthedocs.io/en/latest/MQTT_API_REFERENCE.html)

* https://github.com/custom-components/pyscript

mmattel, 2024-02-29:
> last year i'v done some testing with appdaemon and pyscript concluding a seperate rpi4 zero with custom code works in no time without headaches.

@josy1024 then I am happy reading your post about a working script usage with that 😅

mmattel, 2024-02-29:
Closing via a new release, capable now running in HAOS, see [v4.0.0](https://github.com/mmattel/EET-Solmate/releases/tag/v4.0.0)

closed 2024-02-29 by mmattel