mmattel/EET-Solmate #26: set solmate settings with mqtt

josy1024, 2024-02-28:
> Is there a possibility to add an adjustment for {client.set_min_injection(200)} , {client.set_max_injection(800)} and minimal battery load - sorry I am not capable to write this.

@lukashunziker99 I've running my custom code, because i want to dynamically set min, max and battery_percentage with mqtt.

the sol2mqtt.py script proxys the values to mqtt and subscribes from mqtt,
i'm using node-red integration in home-assistant for changeing the mqtt values.

config-data flows as follows:
* **node-red** function (checks grid values)
** if power flows to grid => load battery (min, max = 0, battery percentage = 95)
** if power flows from grid => use battery (battery_percentage = 20, min, max depends on grid usage between 50-200 watts)
* * => if something changed => write to "subscribe_topics" MQTT
* **sol2mqtt.py** subscribed "subscribe_topics" from MQTT => writes back with official "solclient" (solmate_sdk)

It's working, long-time tested and robust although solmate api is not very stable and the script needs resets every 1-2 hours.
maybe @mmattel could integrate this and use some of my code after ensureing your coding standards.
i could also do some tests in my HAOS installation if you want? 

just need the https://github.com/eet-energy/solmate-sdk running in your env, then you can use this script.
(

https://github.com/josy1024/eet2mqtt/blob/main/sol2mqtt.py
searchfor "subscribe_topics" and "mqtt2sol":
```py

subscribe_topics = ["eet/solmate/0/set/user_maximum_injection", "eet/solmate/0/set/user_minimum_injection", "eet/solmate/0/set/user_minimum_battery_percentage"]

def mqtt2sol(topic, received_message):
    print(f"mqtt2sol: Received message on topic solclient.set {topic}: {received_message}")
    if "user_maximum_injection" in topic:
        solclient.set_max_injection(int(received_message))
    elif "user_minimum_injection" in topic:
        solclient.set_min_injection(int(received_message))
    elif "user_minimum_battery_percentage" in topic:
        solclient.set_min_battery_percentage(int(received_message))
```

_Originally posted by @josy1024 in https://github.com/mmattel/EET-Solmate/issues/25#issuecomment-1968217511_

mmattel, 2024-03-11:
@josy1024 I am working on integrating the three settings as changable values via
HA -> MQTT -> my-code --> Solmate. This takes a bit, pls be patient 😅

mmattel, 2024-04-29:
Done via: https://github.com/mmattel/EET-Solmate/releases/tag/v5.0.0 😃

closed 2024-04-29 by mmattel