mmattel/EET-Solmate #27: mqtt connection failed: [Error -5] Name has no usable address

DarkWingRo, 2024-02-28:
Hello,

I upgraded to HA Core V 2024.2.5 and to the EET-Somate skript version 3.0 and since then I see the following error if I try to start the script: mqtt connection failed: [Error -5] Name has no usable address
[image]

I don't how to resolve this and need help...

thx,
Robert

mmattel, 2024-02-28:
Are you running HA on docker or HAOS?

DarkWingRo, 2024-02-29:
Thank you for your reply... and sorry for the missing information in my first post.

Thanks to your reference to the mosquito broker, I checked the addon and it turned out that it was not set to autostart with HAOS. I just had to start the addon and everything is fine again.

I'close this issue...

thank you

closed 2024-02-29 by DarkWingRo

mmattel, 2024-02-29:
Great to hear 😃
FYI, I just have release v4 with nice improvements, just have a look into the readme...

mmattel, 2024-05-04:
Info:
https://github.com/mmattel/EET-Solmate/releases/tag/v5.0.0
https://github.com/mmattel/EET-Solmate/discussions