mmattel/EET-Solmate #30: V4 does not start Solmate after reboot

lukashunziker99, 2024-03-02:
Thank you very much. I tried it. Sadly it does not work. I think one reason is that there is no subfolder in /config you can open with /….. You have to open it with cd ….. The cd /config/shell/solmate does not work.
The automation actually starts after reboot but does not load/start SolMate  
[image]
[image]

mmattel, 2024-03-02:
hmmmm, can you print me the output of:
ls -la /
and
ls -la /config

lukashunziker99, 2024-03-02:
[image]

mmattel, 2024-03-02:
Thanks 👍
can you also post the content of the homeassistant direcectory?
(sorry for asking these things step by step...)

lukashunziker99, 2024-03-02:
[image]

mmattel, 2024-03-02:
thx, this dir list looks very much the same as mine. what happens if you create the solmate/ directory here or under shell/ incl the setup procedure and define the shell command using the path accordingly. notice that I had to restart HA so that any change on sc gets recognizefd.

lukashunziker99, 2024-03-02:
I created a directory mkdir /homeassistant/solmate and followed your steps. Still it does not work - Solmate is not installed after reboot - I adjusted the configuraton.yaml and the Shell command in automation. I completly restartet HA several times.
Following response after testing crond: 
python crond-execute:
File „/homeassistant/solmate/crond-execute“, line 12
 source „${SCRIPT_DIR}/bin/activate“
             ^^^^^^^^^^^^^^^^^^^^^^

AND
Python crond-prepare:
File „/homeassistant/solmate/crond-prepare“, line 7
 SCRIPT_DIR=„$(dirname „$(readline -f „$0“)“)“
SyntaxError: Invalide syntax

mmattel, 2024-03-03:
There seems to be a misunderstanding somewhere we need to fiure out...
(maybe just a docs thing)

* I have installed HAOS as virtual machine successfully and started it up.
* Then logged into HAOS
* Then logged into their HA docker with `docker exec -it homeassistant bash`
* Created the directories and added a Python venv
* Checked that a library IS NOT present with `python -m pip freeze | grep termcolor`
* Installed that library with `python -m pip install termcolor`
* Rebooted and everything was there
[image]

[image]

mmattel, 2024-03-03:
I think I found it 😄 

The issue lays in the terminal you are using.
You use `Advanced SSH & Web Terminal` via a HA addon which restricts doing stuff, especially changes are not reboot persistent - as you have recognized.

You need to go into the HAOS from `outside` of HA and do the stuff there. It will work then.
The nice thing is, that the HA supervisor has git included so cloning is not an issue.
Note that I was always using a native terminal so this did not came up to my mind.

I will update the documentation accordingly.

lukashunziker99, 2024-03-03:
Thank you. I was able to perform your steps after deactivating Advanced SSH & Web Terminal protected mode.
[image]
However after reboot Solmate was not startet.

mmattel, 2024-03-03:
As said, do not use the Python based terminal emulation at all. You must use the native terminal from HAOS via screen or like with putty (windows) or ssh (linux). I will update the docs accordingly.

mmattel, 2024-03-03:
Closing,
fixed with: https://github.com/mmattel/EET-Solmate/releases/tag/v4.0.1 which is a doc change only.

closed 2024-03-03 by mmattel