mmattel/EET-Solmate #32: Broker Continuously Closing and Reconnecting Connections

JoniXTech, 2024-03-03:
**Issue:** Broker Continuously Closing and Reconnecting Connections

**Description:**
The MQTT broker consistently closes existing connections and establishes new ones, leading to frequent connection disruptions. This behavior cause stuff like that.

https://github.com/mmattel/EET-Solmate/assets/91024155/8f1d7d60-4c84-46c2-9455-89e0d84e8abc

**Observations:**
Upon reviewing the broker logs, it's evident that the following sequence occurs repeatedly:

```
2024-03-03 22:11:00: New connection from 172.30.32.1:48771 on port 1883.
2024-03-03 22:11:00: Client solmate_mqtt already connected, closing old connection.
2024-03-03 22:11:00: New client connected from 172.30.32.1:48771 as solmate_mqtt (p5, c0, k70, u'homeassistant').
2024-03-03 22:11:00: New connection from 172.30.32.1:58851 on port 1883.
2024-03-03 22:11:00: Client solmate_mqtt already connected, closing old connection.
2024-03-03 22:11:00: New client connected from 172.30.32.1:58851 as solmate_mqtt (p5, c0, k70, u'homeassistant').
2024-03-03 22:11:00: New connection from 172.30.32.1:50325 on port 1883.
2024-03-03 22:11:00: Client solmate_mqtt already connected, closing old connection.
2024-03-03 22:11:00: New client connected from 172.30.32.1:50325 as solmate_mqtt (p5, c0, k70, u'homeassistant').
2024-03-03 22:11:00: New connection from 172.30.32.1:39711 on port 1883.
2024-03-03 22:11:00: Client solmate_mqtt already connected, closing old connection.
2024-03-03 22:11:00: New client connected from 172.30.32.1:39711 as solmate_mqtt (p5, c0, k70, u'homeassistant').
2024-03-03 22:11:00: New connection from 172.30.32.1:39129 on port 1883.
2024-03-03 22:11:00: Client solmate_mqtt already connected, closing old connection.
2024-03-03 22:11:00: New client connected from 172.30.32.1:39129 as solmate_mqtt (p5, c0, k70, u'homeassistant').
```

This pattern indicates that the broker is terminating existing connections ("Client solmate_mqtt already connected, closing old connection") and then establishing new connections for the same client.

**Configuration:**
- MQTT Device/Service Settings:
[image]
- `.env` File Configuration:
[image]

**Expected Behavior:**
Connections should be maintained without constant closing and reopening.

mmattel, 2024-03-05:
This comes from that the keep-alive ping/pong gets out of sync which runs internally between mosquitto and paho-mqtt. I occasionally see this on my instance once a day too but only for a view seconds. I dig into the possibilities deeper and see what I can do.

mmattel, 2024-03-08:
As stated in #35, I will provide a patch release asap.

closed 2024-03-08 by mmattel

mmattel, 2024-05-04:
Info: 
https://github.com/mmattel/EET-Solmate/releases/tag/v5.0.0
https://github.com/mmattel/EET-Solmate/discussions

PR #35 (2024-03-08): MQTT: remove keepalive parameter on connect
Fixes: #32 

There is no need to set the keepalive parameter in the connection setup which is especially true for MQTTv5. If set, there can be occurences during the day where you see "flickering" solmate values on the screen.

It also avoids the following messages consectutive getting logged in the broker:
```
Client solmate_mqtt closed its connection.
Client xyz closed its connection.
```

I will provide a patch release asap.