mmattel/EET-Solmate #38: SSH und Basics

schoenis, 2024-03-23:
LIeber Martin, danke für deine schnellen Responses - eröffne hiermit einen Thread:

Ich bin soweit einmal gekommen, dass ich einen SSH Key erstellt habe und diesen auf einen USB STick in meinen Raspberry Pi auf dem HAOS läuft gesteckt.
Dann hab ich versuch mich mit dem HAOS auf meinem iMac zu verbinden.
Das klappt nicht wirklich: bekomme immer wieder connection refused

hab mich an die Anleitung https://www.simon42.com/home-assistant-ssh-root-os gehalten...

schoenis, 2024-03-23:
hab auch schon neu gestartet, ha os import versucht...
hab die Datei auch entsprechend umbenannt bzw auch .pub weggenommen -

schoenis, 2024-03-23:
ok, habs geschafft :D

schoenis, 2024-03-23:
[image]
soweit bin ich gekommen...bein configure env steig ich aus...

mmattel, 2024-03-23:
Hallo @schoenis 👋 

Jetzt kommen erstmal Fragen 😄 

* Wo lässt Du mein Programm laufen:
  * Auf einem externen Server, also nicht innerhalb von HA
  * Innerhalb von HAOS
* Bist Du alle Requirements für die notwendigen Module durchgegangen - es fehlen also keine?
* Kannst Du mir Deine derzeitige `.env` config posten aber OHNE sensible Daten (einfach Sternchen machen)

schoenis, 2024-03-23:
Innerhalb von HAOS auf einem Raspberry pi

solmate) homeassistant:/config/shell/solmate# python check_requirements.py
#
The following modules can be commented in requirements.txt because they are already available:

['asyncio', 'datetime', 'schedule', 'semver', 'setuptools', 'termcolor', 'websockets']

When done, call: python -m pip install -r requirements.txt

If the call returns an empty array like [], all module requirements are met.

Check upgradeable packages with: python -m pip list --outdated
and upgrade them selectively with: python -m pip install --upgrade <package-name>

(solmate) homeassistant:/config/shell/solmate# #

schoenis, 2024-03-23:
# mqtt config
mqtt_server=<core-mosquitto>
mqtt_port=1883
mqtt_username=<homeassistant>
mqtt_password=<xxxxx>
mqtt_client_id=solmate_mqtt
mqtt_topic=solmate
mqtt_prefix=eet
mqtt_ha=homeassistant

eet_local_subdomain="sun2plug"

eet_server_uri=ws://192.168.101.6:9124/

eet_serial_number="<xxxxxxx>"
eet_password="<xxxxxxx>"
eet_device_id="<solmate>"

# timer in seconds
timer_offline=600
timer_conn_err=10
timer_live=30
timer_reboot=180

# timer as number
# when there are too many consecutive response errors, restart after n attempts
# this gives a total waiting time of: timer_live [s] x timer_attempt_restart
# 3 * 30 = 90s = 1.5min
timer_attempt_restart=3

# general envvars configuring program behaviour

# log timer calls. set to false if all works and you just loop thru the live values
# to avoid polluting syslog with data  
general_add_log=False

# enable/disable printing _response_ data to the console, useful for testing
general_print_response=False

# print logging data to console (conditional) and syslog (always)
# the value is optional and defaults to False
general_console_print='True'

# globally enable/disable mqtt, makes it easier for testing
general_use_mqtt=True

schoenis, 2024-03-23:
Habs auch mit wss://sol.eet.energy:9124/ bzw ws://sun2plug.local:9124/ versucht....

schoenis, 2024-03-23:
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# 
(solmate) homeassistant:/config/shell/solmate# python check_reqirements.py
python: can't open file '/config/shell/solmate/check_reqirements.py': [Errno 2] No such file or directory
(solmate) homeassistant:/config/shell/solmate# python check_requirements.py
#
The following modules can be commented in requirements.txt because they are already available:

['asyncio', 'datetime', 'schedule', 'semver', 'setuptools', 'termcolor', 'websockets']

When done, call: python -m pip install -r requirements.txt

If the call returns an empty array like [], all module requirements are met.

Check upgradeable packages with: python -m pip list --outdated
and upgrade them selectively with: python -m pip install --upgrade <package-name>

(solmate) homeassistant:/config/shell/solmate# python check_reqirements.pyy
python: can't open file '/config/shell/solmate/check_reqirements.pyy': [Errno 2] No such file or directory
(solmate) homeassistant:/config/shell/solmate# python check_requirements.py

The following modules can be commented in requirements.txt because they are already available:

['asyncio', 'datetime', 'schedule', 'semver', 'setuptools', 'termcolor', 'websockets']

When done, call: python -m pip install -r requirements.txt

If the call returns an empty array like [], all module requirements are met.

Check upgradeable packages with: python -m pip list --outdated
and upgrade them selectively with: python -m pip install --upgrade <package-name>

(solmate) homeassistant:/config/shell/solmate# python -m pip list --outdated
(solmate) homeassistant:/config/shell/solmate# python -m pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Requirement already satisfied: asyncio in ./lib/python3.12/site-packages (from -r requirements.txt (line 6)) (3.4.3)
Requirement already satisfied: datetime in ./lib/python3.12/site-packages (from -r requirements.txt (line 8)) (5.5)
Requirement already satisfied: paho-mqtt in ./lib/python3.12/site-packages (from -r requirements.txt (line 14)) (2.0.0)
Requirement already satisfied: python-dotenv in ./lib/python3.12/site-packages (from -r requirements.txt (line 15)) (1.0.1)
Requirement already satisfied: semver in ./lib/python3.12/site-packages (from -r requirements.txt (line 16)) (3.0.2)
Requirement already satisfied: setuptools in ./lib/python3.12/site-packages (from -r requirements.txt (line 17)) (69.2.0)
Requirement already satisfied: schedule in ./lib/python3.12/site-packages (from -r requirements.txt (line 18)) (1.2.1)
Requirement already satisfied: termcolor in ./lib/python3.12/site-packages (from -r requirements.txt (line 22)) (2.4.0)
Requirement already satisfied: websockets in ./lib/python3.12/site-packages (from -r requirements.txt (line 24)) (12.0)
Requirement already satisfied: zope.interface in ./lib/python3.12/site-packages (from datetime->-r requirements.txt (line 8)) (6.2)
Requirement already satisfied: pytz in ./lib/python3.12/site-packages (from datetime->-r requirements.txt (line 8)) (2024.1)
(solmate) homeassistant:/config/shell/solmate#

mmattel, 2024-03-23:
Also Dein `.env` schaut gut aus. Du hast Local Access eingestellt. Passt. (der ist immer mit ws:)
Zur Sicherheit, Du kannst Dich per Browser auf deinem Solmate anmelden...

Die Frage ob Du das auf einem eigenen Server oder in HAOS laufen brauch ich noch. Ich gehe stark von HAOS ggf sogar mit einer vorgefertigten Box, also Plug and Play aus. Ich brauche alles diese Info um weitere fragen gezielt stellen zu können. Wenn Du da was überspringst tu ich mir schwer...

(Leider liegt die Komplexität mit HAOS nicht bei mir...)

schoenis, 2024-03-23:
was meinst du mit eigenem Server? Auf meinem raspberry läuft das HA OS, der mittels Ethernet an meinem Router hängt...

mmattel, 2024-03-23:
* Auf meinem raspberry läuft das HA OS...
ok, das hilft schon mal als Info, thx
* über den Browser lässt sich der Solmate ansteuern
Sehr gut, passt also auch
* Die IP ist etwas anders 192.168.101.6, alle anderen Devices laufen in 192.168.50.xx
Bekommst Du von der HAOS shell von wo Du das Script installierst einen positiven Response wenn Du `ping 192.168.101.6` ausführts?
* Hast du wie in der Anleitung angegeben ein Python virtual environment eingerichtet? Ich hab im Shell output oben keinen Hinweis dafür gesehen

schoenis, 2024-03-23:
homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# ping 192.168.101.6
PING 192.168.101.6 (192.168.101.6): 56 data bytes
64 bytes from 192.168.101.6: seq=0 ttl=63 time=33.329 ms
64 bytes from 192.168.101.6: seq=1 ttl=63 time=3.394 ms
64 bytes from 192.168.101.6: seq=2 ttl=63 time=8.709 ms
64 bytes from 192.168.101.6: seq=3 ttl=63 time=3.098 ms
64 bytes from 192.168.101.6: seq=4 ttl=63 time=3.529 ms
64 bytes from 192.168.101.6: seq=5 ttl=63 time=4.380 ms
64 bytes from 192.168.101.6: seq=6 ttl=63 time=4.731 ms
64 bytes from 192.168.101.6: seq=7 ttl=63 time=45.640 ms
64 bytes from 192.168.101.6: seq=8 ttl=63 time=3.083 ms

mmattel, 2024-03-23:
* Ping passt
* Ja den Command, und Du hast danach die venv aktiviert - und DANN die Requirements gemacht.
* Mach mal auch im Verzeichnis wo Du das Script installiert hast: `nslookup 192.168.101.6` (das hat mit dem eigentlichen Problem nichts zu tun, hilft aber für einen Tip)

schoenis, 2024-03-23:
# docker exec -it homeassistant bash
homeassistant:/config# docker exec -it homeassistant bash
bash: docker: command not found
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# nslookup 192.168.101.6
6.101.168.192.in-addr.arpa	name = sun2plug.

(solmate) homeassistant:/config/shell/solmate#

mmattel, 2024-03-23:
Tips:
* mach mal einen `ping sun2plug` (ggf mit dem punkt am Ende)
Wenn das funktioniert kannst Du das statt der IP Adresse `192.168.101.6` in der Konfig eingeben. das hat den Vorteil dass Du Dir nicht die Adresse merken musst denn die kann sich verändern...
* wenn du in GitHub etwas als Block darstellen willst, dann mach am Anfang und am Ende drei Backticks ( ``` ) in je eine eigene Zeile 😄 
* Mit single Backticks kannst Du in einer normalen Zeile ein Wort grau hervorheben

So und nun ein echter Test, davon bräuchte ich den Output, in der venv gib folgendes ein:
`python solmate.py` Bei mir sieht das so aus:
```
Initializing websocket connection.
Create Socket.
Websocket is connected to: ws://sun2plug.xxxxx:9124/
Authenticating.
Authentication successful!
SolMate is online.
Initializing the MQTT class.
Initializing the MQTT client.
MQTT is connected and running.
Update MQTT topics for Homeassistant.
Once a day queries called by scheduler.
```

schoenis, 2024-03-23:
´´´(solmate) homeassistant:/config/shell/solmate# ping sun2plug 
ping: bad address 'sun2plug'
(solmate) homeassistant:/config/shell/solmate# ping sun2plug. 
ping: bad address 'sun2plug.'
(solmate) homeassistant:/config/shell/solmate# ´´´

mmattel, 2024-03-23:
Ich hab da eine Vermutung...
Gib bitte in der Konfig bei `eet_local_subdomain` folgendes ein:
`eet_local_subdomain="192.168.101.6"`

schoenis, 2024-03-23:
wenn ich den Befehl source /config/shell/solmate/bin/activate auch noch eingeben kommt folgendes:

```# docker exec -it homeassistant bash
homeassistant:/config# 
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
source /config/shell/solmate/bin/activate

homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# 
(solmate) homeassistant:/config/shell/solmate# python solmate.py
Initializing websocket connection.
Create Socket.
Websocket is connected to: ws://192.168.101.6:9124/
Authenticating.
Response: {'type': 'IncorrectSerialNumber', 'message': 'This is not the Solmate you were looking for!'}
Authentication failed!
Failed creating connection/authentication to websocket class.
Waiting: timer_offline: 600s

mmattel, 2024-03-23:
```
homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
source /config/shell/solmate/bin/activate
```
Du must die venv nur einmal einrichten.
Aber jedesmal checken ob die venv für das `solmate` verzeichnis aktiv ist. Das zeigt sich wenn Du im solmate Verzeichnis (`cd /config/shell/solmate`) bist an `(solmate) homeassistant:)` am Anfang. Wenn das nicht ist dann einfach nur `source /config/shell/solmate/bin/activate` eingeben, dann ändert sich das sofort.

mmattel, 2024-03-23:
Hast Du schon den Test mit dem geänderten `eet_local_subdomain` gemacht?

schoenis, 2024-03-23:
***# docker exec -it homeassistant bash
homeassistant:/config# 
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# python solmate.py .env
Initializing websocket connection.
Create Socket.
Websocket is connected to: ws://192.168.101.6:9124/
Authenticating.
Authentication failed!
Failed creating connection/authentication to websocket class.
Waiting: timer_offline: 600s

mmattel, 2024-03-23:
Das sind gute Neugigkeiten denn jetzt ist der Fehler eindeutig in der Authentifizierung!
Bitte checke ob Deine Credentials für den Solmate korrekt in `.env` eingegeben sind. Die Credentials müssen in doppelten Anführungszeichen sein, zB:
`eet_serial_number="xyz"`

schoenis, 2024-03-23:
bingo...ein Buchstabe ist verloren gegangen, dafür hab ich MQTT issues - hab da die credentials von meiner MQTT Applikation genommen (Mosquitto broker)
Stimmt das?

```Websocket is connected to: ws://192.168.101.6:9124/
Authenticating.
Authentication successful!
SolMate is online.
Initializing the MQTT class.
Initializing the MQTT client.
MQTT connection failed: [Errno -5] Name has no usable address
(solmate) homeassistant:/config/shell/solmate#

mmattel, 2024-03-23:
So, ich halte folgendes fest 😄 
* Im Wesentlichen ist das mit der `eet_local_subdomain` noch nicht klar. Der Text ist zwar korrekt aber nicht für jeden verständlich. Da muss ich noch nachschärfen.
* Ich frage mich aber wegen dem Output `MQTT is connected and running.` warum das bei dir öfters kommt. Erscheint das im 30s Takt?
* Wenn das Script läuft, solltest Du in HA bei MQTT nun einen Eintrag für den Solmate haben.

schoenis, 2024-03-23:
> * Ich frage mich aber wegen dem Output `MQTT is connected and running.` warum das bei dir öfters kommt. Erscheint das im 30s Takt?

nein im Sekunden Takt - erklärt das das Flackern?

> * Wenn das Script läuft, solltest Du in HA bei MQTT nun einen Eintrag für den Solmate haben.

ja hab einen Eintrag!

mmattel, 2024-03-23:
> nein im Sekunden Takt - erklärt das das Flackern?

Nein, das muß ich noch prüfen. Da ist irgenwas bei HA im Broker denn der schließt die Verbindung andauernd. Kannst Du mal schauen was in den Settings vom Mosquitto in HA ist. Such mal ggf nach einem `keep-alive` Wert.

Wegen crond:
* Check bitte dass die beiden crond-xxx scripte executable sind (siehe Anleitung)
* Beende das python script manuell.
* Starte mit `./cron-prepare` das Script.
Jetzt warte mal so 2-5min und schaue zwischendurch mit dem `ps` Kommando ob sich das python solmate script startet. Meld Dich mit einem (erfolgs) Output.

schoenis, 2024-03-23:
ja aber wie checke ich das?
und wie beende ich das manuell?

mmattel, 2024-03-23:
> ja aber wie checke ich das?
und wie beende ich das manuell?

* Python Scripte die Du von der Kommandozeile gestartet hast kannst Du ganz normal mit `ctrl-c` beenden.
* Ohne es zu prüfen, kannst Du sie mit `chmod +x <script-name>` ändern (+x = executable).

* Nachdem Du `nslookup 192.168.101.6` eingegeben hast, kam ja als name `sun2plug.`
Probier mal `nslookup sun2plug`, also die umgekehrte Variante. Was kommt da?

mmattel, 2024-03-23:
> und wenn alles läuft den SSH kanal wieder schließen indem ich den USB STick abziehe?

Zuerst müssen (wir) den Start mit crond-prepare checken. Wenn das passt und das sollte es, dann kann man den abziehen weil man ja dann die Shell nicht mehr braucht und das Script mit Autostart (reboot proof) starten sollte. Außer man möchte was am Code ggf an den Settings ändern.

schoenis, 2024-03-24:
ok, flackern hat sich gelegt..
welchen Befehl muss ich eingeben um zu überprüfen, dass die 2 crond executable sind?

Kann man das Intervall, mit dem die pv power ausgelesen wird anpassen? (von soweit ich das verstehe 30s auf zb sekündlich?)

mmattel, 2024-03-24:
> ok, flackern hat sich gelegt..

Sehr gut. Ich hab gestern noch geschaut wie ich eine Ausgabe machen kann warum ein Disconnect passiert. Geht bei mir recht gut, muss ich noch veröfentlichen.

> welchen Befehl muss ich eingeben um zu überprüfen, dass die 2 crond executable sind?

* [mit dem 'ls' Kommando](https://www.google.com/search?q=how+to+check+if+a+file+is+executable+in+linux+with+ls+&client=opera&hs=9ZN&sca_esv=fcf9c3ed84842f36&ei=DuP_ZfTPCeTixc8PtNCoiAc&ved=0ahUKEwj0jtuyvIyFAxVkcfEDHTQoCnEQ4dUDCBA&uact=5&oq=how+to+check+if+a+file+is+executable+in+linux+with+ls+&gs_lp=Egxnd3Mtd2l6LXNlcnAiNmhvdyB0byBjaGVjayBpZiBhIGZpbGUgaXMgZXhlY3V0YWJsZSBpbiBsaW51eCB3aXRoIGxzIDIFECEYoAFIlBtQvgdY1BdwAXgBkAEAmAGnAaABvwmqAQQxMC4zuAEDyAEA-AEBmAIJoAKpBcICChAAGEcY1gQYsAPCAggQIRigARjDBMICChAhGAoYoAEYwwTCAgQQIRgVwgIIEAAYgAQYogSYAwCIBgGQBgiSBwE5oAfTJA&sclient=gws-wiz-serp) siehe Dr. Google damit Du den Output lesen kannst.

* Ohne es zu prüfen, kannst Du sie jederzeit mit `chmod +x <script-name>` ändern (+x = executable). (ohne die spitzen klammern). Wenn es schon war, macht es nichts.

> von soweit ich das verstehe 30s auf zb sekündlich

Mach das nicht. Sekündlich bringt Dir gar nichts und "überfordert" das System. Wenn Du später Werte aggregieren möchtest, weil EET das nicht eingebaut hat, musst Du auf die Werte ein Riemann Integral ansetzen. dazu gibt es in HA einen Helper, siehe meine Doku für einen Link.

> das ist der Output von nslookup

Das ist blöd. Du musst (vorerst) bei IP bleiben. Das hat mit dem DNS zu tun aber ich weis noch nicht wo das bei Dir hängt.

mmattel, 2024-03-24:
DNS Nachtrag. Gib mal bitte auf der Shell folgendes ein und poste den Output: `ha dns info`

schoenis, 2024-03-24:
da bekomm ich diesen output:

```# docker exec -it homeassistant bash
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# resolvectl query sun2plug
bash: resolvectl: command not found
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# resolvectl query sun2plug
bash: resolvectl: command not found
(solmate) homeassistant:/config/shell/solmate# resolvect query sun2plug
bash: resolvect: command not found
(solmate) homeassistant:/config/shell/solmate# resolvectl query sun2plug
bash: resolvectl: command not found
(solmate) homeassistant:/config/shell/solmate#

mmattel, 2024-03-24:
ok, mein Fehler... hab vergessen anzugeben von welcher shell 🤦‍♂️ 
Das Kommando musst du _außerhalb_ des docker containers machen. Innerhalb gibt es das nicht - das hast Du gemerkt. Du erkennst es wo Du bist am Shell Prompt. 
* In Docker ist es zb `homeassistant` 
* außerhalb ist es `#` 
* und wenn Du da auch aussteigst bist Du in der HAOS Kommando Konsole die mit `ha` promptet

`ha` --> login --> `#` --> docker exec ... --> `homeassistant`
`homeassistant` --> ctrl-d --> `#` --> ctrl-d --> `ha`

Also mach das Kommando nochmals in `#`

schoenis, 2024-03-24:
```Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# ha

The Home Assistant CLI is a small and simple command line utility that allows
you to control and configure different aspects of Home Assistant

Usage:
  ha [command]

Available Commands:
  addons            Install, update, remove and configure Home Assistant add-ons
  audio             Audio device handling.
  authentication    Authentication for Home Assistant users.
  available-updates Provides information about current pending updates
  backups           Create, restore and remove backups
  banner            Prints the CLI Home Assistant banner along with some useful information
  cli               Get information, update or configure the Home Assistant cli backend
  completion        Generate the autocompletion script for the specified shell
  core              Provides control of the Home Assistant Core
  dns               Get information, update or configure the Home Assistant DNS server
  docker            Docker backend specific for info and OCI configuration
  hardware          Provides hardware information about your system
  help              Help about any command
  host              Control the host/system that Home Assistant is running on
  info              Provides a general Home Assistant information overview
  jobs              Get information and manage running jobs
  mounts            Get information, update or configure mounts in Supervisor
  multicast         Get information, update or configure the Home Assistant Multicast
  network           Network specific for updating, info and configuration imports
  observer          Get information, update or configure the Home Assistant observer
  os                Operating System specific for updating, info and configuration imports
  refresh-updates   Reload stores and version information
  resolution        Resolution center of Supervisor, show issues and suggest solutions
  security          Get information and manage security functionality
  store             Install and update Home Assistant add-ons and manage stores
  supervisor        Monitor, control and configure the Home Assistant Supervisor

Flags:
      --api-token string   Home Assistant Supervisor API token
      --config string      Optional config file (default is $HOME/.homeassistant.yaml)
      --endpoint string    Endpoint for Home Assistant Supervisor (default is 'supervisor')
  -h, --help               help for ha
      --log-level string   Log level (defaults to Warn)
      --no-progress        Disable the progress spinner
      --raw-json           Output raw JSON from the API

Use "ha [command] --help" for more information about a command.
# docker exec -it homeassistant bash
homeassistant:/config# 
exit
# resolvectl query sun2plug
sun2plug: 'sun2plug' not found
# docker exec -it homeassistant bash
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# 
exit
# resolvectl query sun2plug

sun2plug: 'sun2plug' not found
# 
# 
#

mmattel, 2024-03-24:
hmmm 🤔
`resolvectl query sun2plug.` also mit punkt am Ende?
`resolvectl query sun2plug.localdomain`
ich hab gelesen dass es Probleme mit lokalen dns Auflösungen gibt, deshalb das Probieren der Varianten...

schoenis, 2024-03-24:
```# resolvectl query sun2plug.
sun2plug.: 'sun2plug.' not found
# resolvectl query sun2plug.localdomain
sun2plug.localdomain: Name 'sun2plug.localdomain' not found
#

schoenis, 2024-03-24:
- DNS Nachtrag. Gib mal bitte auf der Shell folgendes ein und poste den Output: ha dns info

bin da offenbar schon wieder falsch - bitte schrieb mir immer genau dazu was ich genau davor eingeben sol...

```homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# ha dns info
bash: ha: command not found
(solmate) homeassistant:/config/shell/solmate# python -m venv /config/shell/solmate
(solmate) homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# ha dns info
bash: ha: command not found
(solmate) homeassistant:/config/shell/solmate#

mmattel, 2024-03-24:
Ich meld mich später --> Family

mmattel, 2024-03-24:
bin (kurz wieder da), lass uns schnell am crond weiterarbeiten.
konntest du die beiden scripte auf executable stellen?
kannst du das folgendes script in docker im /config/shell/solmate starten ? venv ist nicht erforderlich da das script das macht `./crond-execute`. wenn das geht sollte die shell gleich wieder da sein, das solmate script gestartet, siehst du in HA. und du kannst mit dem kommando `ps` schauen ob es in der liste der prozesse ist, da sollte dann ein eintrag sein.

schoenis, 2024-03-24:
```# docker exec -it homeassistant bash
homeassistant:/config# 
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# source /config/shell/solmate/bin/activate
(solmate) homeassistant:/config/shell/solmate# ./crond-execute
(solmate) homeassistant:/config/shell/solmate# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     28:25 python3 -m homeassistant --config /config
  166 root      1:32 python -u /config/shell/solmate/solmate.py /config/shell/s
  167 root      1:32 python -u /config/shell/solmate/solmate.py /config/shell/s
  222 root      0:00 bash
  237 root      0:00 bash
  244 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  248 root      0:00 ps
(solmate) homeassistant:/config/shell/solmate#

schoenis, 2024-03-24:
oder so - hier leuchtet crond grün :)

```(solmate) homeassistant:/config/shell/solmate# ls ./crond-execute
./crond-execute
(solmate) homeassistant:/config/shell/solmate# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     28:38 python3 -m homeassistant --config /config
  166 root      1:32 python -u /config/shell/solmate/solmate.py /config/shell/s
  167 root      1:32 python -u /config/shell/solmate/solmate.py /config/shell/s
  222 root      0:00 bash
  237 root      0:00 bash
  244 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  252 root      0:00 ps
(solmate) homeassistant:/config/shell/solmate#

schoenis, 2024-03-24:
das mit dem kill hab ich auch noch nicht geschafft - wenn ich in der HAOS Terminal `ps eingebe sehe ich das solmate.py nicht in der Liste, obwohl es über MQTT (trotz neuerlich aufgetretenem Flackern) sehe.

mmattel, 2024-03-24:
also erst einmal congrats, es läuft prinzipiell... 😄 
nun ein paar infos in der `homeassistant` shell (da wo du in docker bis wo das /config/... verzeichnis ist) gibst du das kommando `ps` ein. dann siehst di die liste wiedu oben gepostet hast. jetzt kommt der punkt dass du schauen musst dass du nur ein `solmate.py` am laufen hast. deswegen kille alle bis auf einen mit `kill -9 nummer`. die nummer ist die zahl die vor dem root steht. bitte nichts anderes killen...

hinweis: ich könnte natürlich das killen automatisieren, was ist wenn aber jemand mehrere solmates einbindet, dann killt das alles was nicht gewünscht ist...

schoenis, 2024-03-24:
```
# docker exec -it homeassistant bash
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     30:06 python3 -m homeassistant --config /config
  166 root      1:37 python -u /config/shell/solmate/solmate.py /config/shell/s
  167 root      1:37 python -u /config/shell/solmate/solmate.py /config/shell/s
  222 root      0:00 bash
  244 root      0:05 python -u /config/shell/solmate/solmate.py /config/shell/s
  254 root      0:00 bash
  260 root      0:00 ps
homeassistant:/config# kill -9 166
homeassistant:/config# kill -9 167
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     30:11 python3 -m homeassistant --config /config
  222 root      0:00 bash
  244 root      0:06 python -u /config/shell/solmate/solmate.py /config/shell/s
  254 root      0:00 bash
  261 root      0:00 ps
homeassistant:/config#

mmattel, 2024-03-24:
es geht  alles und du nur noch einen solmate.py prozess am laufen,da bist jetzt schon echt weit und es fehlt nur noch der automatisierungs schritt.bei dem gehst du bitte wie in der beschreibung [HA Stuff](https://github.com/mmattel/EET-Solmate/blob/main/docs/prep-ha.md#ha-stuff) vor.

wichtig ist, dass du **vor** dem _test_ den letzten solmate prozess killst den du ja per kommandozeile gestartet hast. da darf keiner am laufen sein.

wenn das auch funktioniert und in HA der solmate anzeigt, kommt der letzte schritt und der ist ein reboot vom HA. dann sollte nach 3-5min der solmate automatisch da sein

schoenis, 2024-03-24:
ja das hab ich sch9on gemacht (Automatisierung + shell command)

mmattel, 2024-03-24:
> dass heißt jetzt kill ich den letzten Prozess

ja, und dann den test schritt. da sollte etwas zeitverzögert dann der solmate in HA auftauchen. du kannst wenn du willst zwischendurch mit ps schauen ob er schon aufgetaucht ist.

schoenis, 2024-03-24:
```homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     31:24 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  295 root      0:00 crond -b -l 3 -L /config/shell/solmate/crond.log -c /etc/c
  297 root      0:00 ps
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     31:26 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  313 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  314 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  319 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  346 root      0:00 ps
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     31:26 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  313 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  314 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  319 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  347 root      0:00 ps
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     31:27 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  313 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  314 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  319 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s

mmattel, 2024-03-24:
jetzt wird es wieder spannend, frage: du hast _alle_ gekillt und dann _einmal_ getestet

schoenis, 2024-03-24:
yes!

mmattel, 2024-03-24:
mac terminal?
du bist im docker von ha, und schaust welche prozesse mit ps da sind, killst alle solmate und da darf dann keiner mehr sein. dann einmal den test über den browser in der HA oberfläche machen, nicht mehrfach.

schoenis, 2024-03-24:
Hab nochmal alle gekillt - jetzt sinds sogar 4

```
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     32:29 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  385 root      0:00 crond -b -l 3 -L /config/shell/solmate/crond.log -c /etc/c
  391 root      0:00 ps
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     32:30 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  412 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  413 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  414 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  421 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  452 root      0:00 ps

schoenis, 2024-03-24:
Hier das ist nachdem ich alle gekillt habe!
```
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root     32:23 python3 -m homeassistant --config /config
  222 root      0:00 bash
  254 root      0:00 bash
  351 root      0:00 ps

mmattel, 2024-03-24:
ich denke ich weis, aber da müssen wir jetzt die hardcore variante machen, a la windows - pls reboot den HA server. dann nochmals in die docker shell einsteigen und beides HA und ps beobachten.

schoenis, 2024-03-24:
```
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root      0:52 python3 -m homeassistant --config /config
   87 root      0:00 bash
  172 root      0:00 crond -b -l 3 -L /config/shell/solmate/crond.log -c /etc/c
  178 root      0:00 ps
homeassistant:/config# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root      0:53 python3 -m homeassistant --config /config
   87 root      0:00 bash
  204 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  205 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  206 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  213 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  214 root      0:00 python -u /config/shell/solmate/solmate.py /config/shell/s
  259 root      0:00 ps
homeassistant:/config#

mmattel, 2024-03-24:
zuerst einmal die gute nachricht. prozess 172 wurde aufgerufen was bedeutet dass der start funktioniert hat und dass er auch korrekt beendet wurde. dass jetzt aber 5 solmate prozesse laufen finde ich "faszinierend"...
was passiert wenn du alle außer den 204 (das ist der älteste) killst? kommen die anderen wieder?

schoenis, 2024-03-24:
et voila

```
Use `ha` to access the Home Assistant CLI.
# docker exec -it homeassistant bash
homeassistant:/config# cd /config/shell/solmate
homeassistant:/config/shell/solmate# ps
PID   USER     TIME  COMMAND
    1 root      0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
   16 root      0:00 s6-supervise s6-linux-init-shutdownd
   18 root      0:00 /package/admin/s6-linux-init/command/s6-linux-init-shutdow
   25 root      0:00 s6-supervise s6rc-fdholder
   26 root      0:00 s6-supervise s6rc-oneshot-runner
   34 root      0:00 /package/admin/s6/command/s6-ipcserverd -1 -- /package/adm
   65 root      0:00 s6-supervise home-assistant
   67 root      1:07 python3 -m homeassistant --config /config
  160 root      0:00 bash
  194 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  196 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  200 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  202 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  204 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  230 root      0:01 python -u /config/shell/solmate/solmate.py /config/shell/s
  253 root      0:00 ps
homeassistant:/config/shell/solmate#

mmattel, 2024-03-24:
kannst du mal
`ls /etc/periodic/solmate/` eingeben und den output posten

schoenis, 2024-03-24:
```
homeassistant:/config/shell/solmate# ls /etc/periodic/solmate/
crond-execute

[16 more replies]

closed 2024-04-02 by mmattel