mmattel/EET-Solmate #39: Typo in documentation

axcro7, 2024-03-24:
https://github.com/mmattel/EET-Solmate/blame/9b06b314cc40cdc5add6d6fc7d2f90b383f29789/docs/prep-ha.md#L71

Following the instructions, I ran into this error wondering what I did wrong:
`python: can't open file '/config/shell/solmate/check_reqirements.py': [Errno 2] No such file or directory`

looks like there is a typo in the documentation
`python check_reqirements.py` -> should be `python check_requirements.py`

PS: thanks for the hard work you've put into this!

mmattel, 2024-03-25:
Nice catch 🤦‍♂️ thanks for highlighting 😄 Will fix asap.

For curiosity, mind to tell how you installed it, on an own server or inside HAOS?

axcro7, 2024-03-25:
You are wellcome ;)

While your're at it, there one more thing I noticed: 
After adding the shell command in the configuration.yaml file, HA needs to be restarted for the command to be available. Reloading the config doesn't seem to be enough.

I installed it inside HAOS (Homeassistant yellow)
It worked yesterday, but during the night (after about 2h) it stopped sending data to HA. I guess I will open a new issue for this.

mmattel, 2024-03-25:
Thanks, good to know.
I am working on an update and I have an idea for how to solve it. The fencing of HAOS is a nightmare...

axcro7, 2024-03-25:
Might have been my fault. I think I hat multiple instances running or something, the values were updated every second and the values were flickering.
Maybe related to issue #40 ?

mmattel, 2024-03-25:
Multiple instances is the thing I am looking for. I am preparing a code change that should prevent that. And this is only for HAOS. Running the script on a normal server is so much easier...

axcro7, 2024-03-26:
This getting off-topic, but have you thought about creating a Docker Image?
If you want, I could look into this.

PS: since cleaning up the `/etc/crontabs/root` file and switching to local connection yesterday, it's working fine! 
Also I believe my account got locked due to too many API queries (can't log in with a browser any more).

mmattel, 2024-03-26:
> have you thought about creating a Docker Image?

Technically, a docker container would ease things, but how you want to integrate an external container in HAOS?

axcro7, 2024-03-26:
I haven't yet tried running containers directly. But from what I have seen, the Add-ons from the HA/HACS store are containers, so there should be a way. 

The documentation looks quite straight forward:
https://developers.home-assistant.io/docs/add-ons/tutorial
I will look into this matter after work, this is something I wanted to try for a while already.

mmattel, 2024-03-31:
@axcro7 mind to test PR #41 which improves the HAOS integration. See the comment for changes. There is btw no breaking change, just fixes and improvements.

I dont know how much you are fine with git, but in a nutshell, you drop any changes that are not automatically excluded like the config, then you checkout master and do a rebase and switch to the branch of 41. Finally you kill all solmate scripts and trigger the automation to start the crond-prepare script. After a while, check that it is running with ps and that there is only one instance of it.

Let me know your findings.

mmattel, 2024-04-02:
Update ist released: https://github.com/mmattel/EET-Solmate/releases/tag/v4.1.0 😃

closed 2024-04-02 by mmattel

axcro7, 2024-07-06:
@mmattel  sorry for not replying earlier :( 
I did look into the container thing, but then failed at including python modules, due to my lac of experience with python..
Then life struck and I didn't get to it any more :/

Anyway, the whole thing is running stable since updating to v4.1.0
I will try to migrate to v6 soon.

mmattel, 2024-07-06:
Appreciate your feedback. Supporting my work is welcomed 😃

Release v6 will be released soon. I want to add one requested functionality with respect to logging and will tag afterwards to make it available.

As you are coming from v4, check the steps v4 -> v5 -> v6 for a smooth upgrade 😃

axcro7, 2024-07-06:
Thanks for the reminder, I will of course look into your documentation / release notes

Not sure if I can support with code, but I will buy you a coffee ☕👍