mmattel/EET-Solmate #43: Solmate neuste Version startet nicht automatisch

lukashunziker99, 2024-04-05:
Hi
Herzlichen Dank Dir. 
Ich habe gemäss Deinen Angaben alles installiert (über putty und root 22222). Ich habe ein HomeAssistant Yellow. Solmate startet leider nicht automatisch:
In nohup.log kommt folgende Meldung: Traceback (most recent call last):
  File "/config/shell/solmate/solmate.py", line 10, in <module>
    import solmate_check as check
  File "/config/shell/solmate/solmate_check.py", line 1, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
Traceback (most recent call last):
  File "/config/shell/solmate/solmate.py", line 10, in <module>
    import solmate_check as check
  File "/config/shell/solmate/solmate_check.py", line 1, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'

Unter python check_requirements.py muss ich pkg_resource nicht installieren - also in requirements.txt mit ## versehen.  Jedoch auch wenn ich pkg_resource nicht mit ## markiere, startet es nicht und "No module named pkg_resources" kommt im nohup.log.
Mit pip -m install -r requirements.py ohne ## in requirement.txt kommt folgende Meldung: ERROR: no matching distribution found for pkg_resources.
Danke

mmattel, 2024-04-06:
Hallo @lukashunziker99 
das Package gedöhns mit Python macht mich noch wahnsinnig.

Frage, könntest du ggf auf der shell im [docker](https://github.com/mmattel/EET-Solmate/blob/main/docs/prep-ha.md#prepare-basic-setup) von HA folgendes [eingeben](https://github.com/mmattel/EET-Solmate/blob/main/docs/prep-ha.md#first-test) und mir den ouptut posten, komplett incl dem shell prompt?

```
cd /config/shell/solmate
source /config/shell/solmate
python -u solmate.py
```

lukashunziker99, 2024-04-06:
[image]

lukashunziker99, 2024-04-06:
[image]
Und über ROOT PORT 22222 im DOCKER wie im Bild oben. Komisch zwei verschiedene Python Versionen.

mmattel, 2024-04-06:
Verflixt, es sollte heissen:

source /config/shell/solmate/bin/activate

Kannst Du den Test nochmals wiederholen...?

lukashunziker99, 2024-04-06:
[image]
[image]

lukashunziker99, 2024-04-06:
[image]
Und hier noch requirements Test mit und ohne ## vor pkg_resources

mmattel, 2024-04-06:
Bin gerade unterwegs und muss mir das genauer anschauen. Die Aussagen der Bilder sind nicht konsistent. Ich meld mich asap.

lukashunziker99, 2024-04-06:
Die mit Welcome to Home Assistant OS sind unter Root über Port 22222 im Docker. Die anderen sind direkt im HA Terminal.

mmattel, 2024-04-06:
Mach mal bitte folgendes:

Im docker (NICHT über das HA terminal!!
```
cd /config/shell/solmate
source /config/shell/solmate/bin/activate
python -m pip install --upgrade pip setuptools
python -m pip install pkg_resources
python -u solmate.py
```
Beim Install der `pkg_resources` könnte nun eine already installed meldug kommen, wenn nicht sollte es installiert werden.

lukashunziker99, 2024-04-06:
[image]
Ja, so funktioniert es - obwohl mit python -m pip install pkg_resources ein ERROR erscheint.

lukashunziker99, 2024-04-06:
[image]
Startet jetzt automatisch - siehe nohup.log.
Danke.
Muss ich python -m pip install --upgrade pip setuptools ggf. bei updates wiederholen?

mmattel, 2024-04-06:
Nein, musst Du nicht weil das ja ein Modul ist das jetzt upgedatet wurde.
Du musst es nur wenn Du das Python venv löscht und neu erstellst.
Ich mach einen dok Eintrag dazu.

Problemhintergrund:
* `pkg_resources` wurde mit 3.12 deprecated. Soweit jetzt nicht das Drama, ich muß einen Ersatz finden. Der Sch*** ist nur, dass der Ersatz bei 3.12 geht aber bei 3.9 muckt, sonst hätte ich das schon früher gemacht.
* Setuptools muß man manuell upgraden da es im Paket von Python 3.12 drinnen ist. In 3.12 ist aber ein altes ("defektes"). Dh, das requirement für setuptools ist gegeben die Version aber veraltet 🤦‍♂️ 

Paketmanagement in Python hat gelinde gesagt noch Potential.

Ich denke schon länger darüber nach das ganze als HA addon zu bauen (in einem eigenen docker container) womit dann diese Probleme weg wären. Die HA doku dazu hat aber auch noch Potential und ich wenig Zeit.

Viel Vergnügen mit der neuen Version 😃 

Wenn alles soweit passt, dann schließe bitte den Issue.

closed 2024-04-06 by mmattel

mmattel, 2024-05-04:
Info: https://github.com/mmattel/EET-Solmate/discussions

PR #44 (2024-04-06): Add docs about required updating setuptools (and pip)
Fixes: #43 

Mandatory update setuptools - added to the docs.