mmattel/EET-Solmate #87: Error: ModuleNotFoundError: No module named 'dotenv'

maikischa, 2024-09-28:
After installing accordance to the instructions I've got the error 
ModuleNotFoundError: No module named 'dotenv'
I added the module photon-dotenv to the modules which is not part in the requirements as a workaround

mmattel, 2024-09-28:
Hmmm,
I dont know from your issue if you are using the plain or Appdaemon installation, but docs clearly state that the modules named in requirements.txt need to be installed.
Can you please double check?

closed 2024-09-29 by mmattel

mmattel, 2024-09-29:
@maikischa as you have seen with the reference PR, I did some doc improvements. Can you drop a note which installation you use for Appdaemon (AddOn/Container)?

maikischa, 2024-09-29:
> 

I used Appdemon as AddOn on a vanilla install of HA2024.9.3 on an x86Arch

PR #88 (2024-09-29): Update Appdaemon docs
Fixes: #87 (Error: ModuleNotFoundError: No module named 'dotenv')

Though there was no code issue and also docs did not miss steps, I took the chance based on the referenced issue to improve the AD docs a bit for more clarity.

@maikischa hope that this makes the steps better understandable and more streamlined.