mmattel/EET-Solmate #89: esham version not correct

BetaTobi, 2024-10-01:
Hi,

I updated to esham version 7.2.0 (at least I thought so...).
And this version is shown in the Terminal.
But the information given by the entities is still on 7.1.0.

See screenshots.

Is this an issue or did I do something wrong?

Best regards,

Torben

[image]

[image]

mmattel, 2024-10-01:
My bad 🤦‍♂️
I missed to change the version string. I will fix this asap. Sorry for the confusion...

mmattel, 2024-10-01:
@BetaTobi see: https://github.com/mmattel/EET-Solmate/releases/tag/v7.2.1

closed 2024-10-01 by mmattel