mmattel/EET-Solmate #94: AppDaemon depricated

gorischek, 2025-03-09:
Hi,
based on https://www.facebook.com/groups/HomeAssistant/posts/3992604054344268/  AppDaemon gets deprocated. What does it mean for this integration?
Br, Michael

mmattel, 2025-03-24:
Hi @gorischek 

thanks for getting in touch and sorry for my delayed reply. I have been quite busy atm.
To answer your question, I try to give you an overview:
* ATM, nothing happens, there is no deprecation announcement on HA though rumor and uncertainty goes around. Frankly speaking, the way this has been started is imho totally nuts. Just my 2c.
* Technical note: an AddOn's are container. Means they are fenced and whatever is inside, it just has to speak the HA API which has no related breaking change. Therefore this deprecation info posted as issue in the AD github page is absolutely questionable, because there is no technical reason behind.
* As far I can tell, without guarantee, any deprecation and removal would finally mean that nothing changes for existing users. AD will not be removed from your installation, only new installation would be affected.
* Consider that any deprecation ONLY affects users who have an HA appliance or installed HA in a way where AddOns are supported.
  * Users who have a containerized environment are NOT affected in any way !! 
* For any possible new user after a written deprecation, I recommend installing AD as AddOn, just to be sure.
* I am in contact with the developer of AD and they are working on a new release. Though I can do any promises, I at least hope that a new version will be available in a timely manner - relaxing the situation. Also see my [post](https://github.com/AppDaemon/appdaemon/issues/2190#issuecomment-2713606598) and the answers in the AD issue.

Hope that this helps, don't hesitate to get in touch.

Note that I have release a new version of `esham` today 😃

gorischek, 2025-03-31:
Hi,
thanks a lot for the feedback!

closed 2025-03-31 by gorischek