mmattel/EET-Solmate #105: AppDaemon

Fox7AT, 2025-05-29:
Servus,

since I updated AppDaemon yesterday from 0.17.1 to 0.17.2 it fails to start solmate.

[12:37:12] INFO: Service AppDaemon exited with code 0 (by signal 0)
[12:37:13] INFO: Starting AppDaemon...
2025-05-29 12:37:17.566825 INFO AppDaemon: ------------------------------------------------------------
2025-05-29 12:37:17.567307 INFO AppDaemon: AppDaemon Version 4.5.3 starting
2025-05-29 12:37:17.567461 INFO AppDaemon: ------------------------------------------------------------
2025-05-29 12:37:17.567642 INFO AppDaemon: Python version is 3.12.10
2025-05-29 12:37:17.568218 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2025-05-29 12:37:17.568489 INFO AppDaemon: Added log: AppDaemon
2025-05-29 12:37:17.568651 INFO AppDaemon: Added log: Error
2025-05-29 12:37:17.568772 INFO AppDaemon: Added log: Access
2025-05-29 12:37:17.568905 INFO AppDaemon: Added log: Diag
2025-05-29 12:37:17.569035 INFO AppDaemon: Added log: Solmate_Log_0
2025-05-29 12:37:17.570000 INFO AppDaemon: Using /config/apps as app_dir
2025-05-29 12:37:17.571290 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module appdaemon.plugins.hass.hassplugin
2025-05-29 12:37:17.571585 INFO HASS: HASS Plugin initialization complete
2025-05-29 12:37:17.576183 INFO AppDaemon: Initializing HTTP
2025-05-29 12:37:17.576441 INFO AppDaemon: Using 'ws' for event stream
2025-05-29 12:37:17.576782 INFO AppDaemon: HTTP Listening on port 5050
2025-05-29 12:37:17.578742 INFO AppDaemon: Starting API
2025-05-29 12:37:17.580329 INFO AppDaemon: Starting Admin Interface
2025-05-29 12:37:17.580614 INFO AppDaemon: Starting Dashboards
2025-05-29 12:37:17.589392 INFO HASS: Connected to Home Assistant 2025.5.3 with aiohttp websocket
2025-05-29 12:37:17.592971 INFO AppDaemon: Starting Apps with 2 workers and 2 pins
2025-05-29 12:37:17.593808 INFO HASS: Authenticated to Home Assistant 2025.5.3
2025-05-29 12:37:17.595496 INFO AppDaemon: Running on port 5050
2025-05-29 12:37:17.595762 INFO AppDaemon: Waiting for plugins to be ready
2025-05-29 12:37:17.596448 INFO HASS: Waiting for Home Assistant to start
2025-05-29 12:37:17.601029 INFO AppDaemon: All plugins ready
2025-05-29 12:37:17.602047 INFO AppDaemon: Scheduler running in realtime
2025-05-29 12:37:17.613151 INFO HASS: Completed initialization in 30.651ms
2025-05-29 12:37:17.723619 INFO AppDaemon: New app config: hello_world
2025-05-29 12:37:17.723846 INFO AppDaemon: New app config: solmate_0
2025-05-29 12:37:17.821650 WARNING AppDaemon: ------------------------------------------------------------
2025-05-29 12:37:17.821864 WARNING AppDaemon: Unexpected error during run()
2025-05-29 12:37:17.822081 WARNING AppDaemon: ------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/appdaemon/__main__.py", line 157, in run
    loop.run_until_complete(asyncio.gather(*pending))
  File "/usr/lib/python3.12/asyncio/base_events.py", line 691, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/appdaemon/utility_loop.py", line 144, in loop
    await self.AD.app_management.check_app_updates(mode=UpdateMode.INIT)
  File "/usr/lib/python3.12/site-packages/appdaemon/app_management.py", line 809, in check_app_updates
    await self._import_modules(update_actions)
  File "/usr/lib/python3.12/site-packages/appdaemon/app_management.py", line 1055, in _import_modules
    load_order = self.dependency_manager.python_sort(modules)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/appdaemon/dependency_manager.py", line 210, in python_sort
    order = [n for n in topo_sort(self.python_deps.dep_graph) if n in modules]
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/appdaemon/dependency.py", line 261, in topo_sort
    raise CircularDependency(f"Visited {visited} already, but {node} depends on {deps}")
appdaemon.dependency.CircularDependency: Visited ['select', 'paho.mqtt.enums', 'enum', 'urllib.request', 'uuid', 'collections', 'paho.mqtt.packettypes', 'solmate_utils', 'sys', 'os', 'solmate_importmanager', 'contextlib', 'importlib', 'subprocess', 'syslog', 'datetime', 'asyncio', 'queue'] already, but solmate_utils depends on {'sys', 'os', 'solmate_importmanager', 'syslog', 'datetime', 'importlib', 'asyncio', 'queue'}
2025-05-29 12:37:17.826363 INFO AppDaemon: Previous message repeated 1 times
2025-05-29 12:37:17.826565 INFO AppDaemon: AppDaemon Exited

gorischek, 2025-05-29:
strange the same error happens without solmate and is also reported several times in appdoamen.
However: I cannot find a 7.3.0:

[image]

What I'm doing wrong?

mmattel, 2025-05-29:
OK, in document [Download and Update esham Using git](https://github.com/mmattel/EET-Solmate/blob/main/docs/download-with-git.md) it as on top the section `Initial Cloning, Commands if you have not cloned esham before` --> you only do to do this one time on the very first setup. Follow the section below `Commands for updates` when you want to update `esham`. The tag list command will show you all the available tags including the newest one you will checkout (make active).

Fox7AT, 2025-05-29:
Servus Martin,
I did not understand, that I had to update manualy. Somehow I remembered, that I don't have to touch it... anyway... It works!

What I did:

- Stopped AppDaemon
- used the terminal in HA to cd to /addons/solmate
- entered: 
- git checkout main
- git stash -u && git stash drop
- git pull --rebase origin main
- git fetch --all --tags
- started the samba addon
- opened the windows explorer with \\HA-IP\addons\solmate
- copied all *.py files
- changed diectory to \\HA-IP\addon_configs\????????_appdaemon\apps\solmate
- pasted the *.py files
- started AppDaemon
- was happy as log showed a starting AppDaemon :)

Thank you!

[https://www.buymeacoffee.com/martin.mattel](url)

closed 2025-05-29 by Fox7AT